olkal/HX711_ADC #5: Works with esp8266?

maxholgasson, 2017-11-24:
Hello,
is this supposed to work with the ESP8266 module?

Get only errors and the module restart and restart

`⸮Wait...

Soft WDT reset

ctx: cont 
sp: 3ffef220 end: 3ffef430 offset: 01b0

>>>stack>>>
3ffef3d0:  3ffe84a0 00000000 3ffee3e4 40202139  
3ffef3e0:  3ffe8425 feefeffe 3ffee2ac 40201ea6  
3ffef3f0:  feefeffe feefeffe 3ffee2ac 40201ed3  
3ffef400:  3fffdad0 3ffee2ac 3ffee3e4 40201c2b  
3ffef410:  feefeffe 00000000 3ffee400 402025f4  
3ffef420:  feefeffe feefeffe 3ffee410 40100114  
<<<stack<<<
⸮⸮b!⸮⸮d
Wait...
`

olkal, 2017-11-24:
I have not tested the library with ESP8266, but it should work. I think the problems you see is related to the ESP pin mapping and pin usage (that I'm not very familiar with myself). I did some tests now with a nodeMCU (ESP12-E), but without any HX711 connected;

This will not work:
//HX711 constructor (dout pin, sck pin)
HX711_ADC LoadCell_1(4, 5); //HX711 1
HX711_ADC LoadCell_2(6, 7); //HX711 2

This seems to work:
//HX711 constructor (dout pin, sck pin)
HX711_ADC LoadCell_1(D2, D1); //HX711 1
HX711_ADC LoadCell_2(D8, D7); //HX711 2

layaannj, 2018-08-14:
hi,

I am also getting the similar issue when connected to nodemcu .
⸮Wait...

Soft WDT reset

ctx: cont
sp: 3ffef220 end: 3ffef430 offset: 01b0

stack>>>
3ffef3d0: 3ffe84a0 00000000 3ffee3e4 40202139
3ffef3e0: 3ffe8425 feefeffe 3ffee2ac 40201ea6
3ffef3f0: feefeffe feefeffe 3ffee2ac 40201ed3
3ffef400: 3fffdad0 3ffee2ac 3ffee3e4 40201c2b
3ffef410: feefeffe 00000000 3ffee400 402025f4
3ffef420: feefeffe feefeffe 3ffee410 40100114
<<<stack<<<
⸮⸮b!⸮⸮d
Wait...
`

how will i connect a 4 wire load cell to nodemcu without using HX711 breakout board???

Thank you

olkal, 2018-08-14:
Hi
This suggests that the wiring between ESP8266 and HX711 is wrong, with an endless loop as a result. Please check your wiring and check correspondence with the assigned pins in your code. 

I will include a fix in the next update of the library to avoid WDT reset of the ESP8266.

You can not use the load cell without the HX711, a microcontroller is not able to read the load cell without an AD converter/amplifer like the HX711.

closed 2018-08-14 by olkal

Gennadwed, 2021-02-28:
//HX711 constructor (dout pin, sck pin)
HX711_ADC LoadCell_1(D2, D1); //HX711 1 здравствуйте ,запускаю так ,ругается на строку " float oldCalibrationValue = LoadCell.getCalFactor();" в начале скетча :
//pins:
const int HX711_dout = 4; //mcu > HX711 dout pin
const int HX711_sck = 5; //mcu > HX711 sck pin

//HX711 constructor:
//HX711_ADC LoadCell(HX711_dout, HX711_sck);
HX711_ADC LoadCell_1(D4, D5); //HX711 1
 как исправить, исползую esp8266?

mdroberts1243, 2022-01-29:
Changing the calibration example  to define:
```
//pins:
const int HX711_dout = D2; //mcu > HX711 dout pin
const int HX711_sck = D1; //mcu > HX711 sck pin
```
Worked great!  There should be a way to put this in the example code in a generic/compatible form... it would save a lot of time!