olkal/HX711_ADC #7: Endless loop, if no HX711 is connected

Captayne, 2017-11-26:
I use the code on my ESP8266 and the tare command doesn't finish, if no Hardware is connected. This results in unwanted watchdog-Resets.
For me the hardware should be optional and recurrent reset is not an option.

My changes: 

void HX711_ADC::tare() 
{
	uint8_t rdy = 0;
	doTare = 1;
	tareTimes = 0;
	int okcounter = 200;
	while(rdy != 2) {
		rdy = update();
		if (okcounter-- == 0){
			 // Prevents from endless loop and watchdog-firing, if no HX711 is connected
			break;
		}
	}
}

olkal, 2017-11-26:
Hi, I see what you mean, and I agree that it should be possible to run the code without HW. However, I think a timeout will be more reliable than a counter. Could you please try the mod below, and let me know if it works for you:

in HX711_ADC.h: (below #define DATA_SET)
```
#define TARE_TIMEOUT (DATA_SET * 100) + 500 // tare timeout time in ms, no of samples * 100ms (10SPS) + 500ms margin
```

in HX711_ADC.cpp:
```
void HX711_ADC::tare() 
{
	uint8_t rdy = 0;
	doTare = 1;
	tareTimes = 0;
	long timeout = millis() + TARE_TIMEOUT;
	while(rdy != 2) {
		rdy = update();
		if (millis() > timeout){ 
			break; // Prevents endless loop if no HX711 is connected
		}
	}
}
```

olkal, 2018-03-09:
Change not implemented.

closed 2018-03-09 by olkal