olkal/HX711_ADC #8: Reading both channels A & B

maxholgasson, 2017-11-29:
Hello,

the HX711 has to channels. I would like to connect two load cells but how can I read them?

olkal, 2017-11-29:
Hi
The library does not support reading from both channels. For two load cells you will need two HX711 modules (the B channel is not as useful as the A channel anyway, see data sheet for details)

maxholgasson, 2017-12-04:
Hi, ok thanks I will do it with a second cell. I want to built a scale for a bee hive and the ESP8266 is solar powered. Therefore I send it to deep sleep and wake it up after a couple of minutes by resetting the module. Is there a way to disable the auto tara function in the lib?

olkal, 2017-12-04:
It is possible to skip auto tare by leaving out the start() function in setup(). 
But there is another issue with your project; the tare point will drift over time and with temperature change. 
In order to get accurate readings over time, my guess is that you will need some clever mechanics to lift the load (hive) or lower the load cells a few mm's, so that you can do a new tare before each reading. This will consume some power, but extending the time between each reading might help. Sleep function for the HX711 is not implemented in the library, but pulling PD_SCK pin high before ESP8266 sleep should do the trick (see data sheet).

closed 2018-03-09 by olkal