olkal/HX711_ADC #12: Change number of samples on smooth average

rafaellehmkuhl, 2018-04-21:
Is there a way to change the number of samples used for the smooth average without changing the library (HX711_ADC.h) itself? As far as I noticed there isn't any function to change this variable.

If there isn't indeed, would you accept it as a Pull Request?

olkal, 2018-04-22:
Thanks, I'm glad you like it!

First of all, the value that decides the number of samples is a definition, not a variable, hence it can not be changed after compilation. 

The scetch and the library files are compiled seperatly, and I'm not aware of any way to pass a value from the scetch that will affect the compilation of the library. You could make a way to pass a value in the scetch to reduce the numper of samples actually used by the library, but as the size of the declared dataset array directly affects memory usage it's not a good ida to declare a larger dataset array than you actually need. 

So the anwer is no, unless someone can come up with a way to do this without making the library more complicated to use, and without using more memory than neccesary.

Olav

jpk73, 2019-05-06:
Hi, I also would be interested in a way to change the number of samples: I am changing the speed on the fly and would like to adjust the number of samples together with the speed change. Would that be possible?

olkal, 2019-05-07:
Hi
Well, yes that should be possible by compiling the full data set and then (on the fly) restricting the library to operate with only a given limited section of the full set. I'm re-opening this Issue. Give me a couple of days and I will look into it.

jpk73, 2019-05-07:
Thanks, that will help a lot!

olkal, 2019-05-09:
Hi!
You should now be able to use the new function setSamplesInUse(int samples) and getSamplesInUse() to manipulate/override the number of samples in use at the time to a lower value than defined in the config.h file.  Note that the memory usage will reflect that of the compiled data set and will not be reduced by using this function.

closed 2019-05-09 by olkal

jpk73, 2019-05-10:
Hi! That helps, thanks a lot! And I can confirm that everything works well on the ADS1231. Just had to do new calibrations because the tare factor changed, but that's no problem at all. I just your old calibration sketch and modified it to give me filtered results, in case you are interested I copy it here:
```
//-------------------------------------------------------------------------------------
// HX711_ADC.h
// Arduino master library for HX711 24-Bit Analog-to-Digital Converter for Weigh Scales
// Olav Kallhovd sept2017
// Tested with      : HX711 asian module on channel A and YZC-133 3kg load cell
// Tested with MCU  : Arduino Nano
//-------------------------------------------------------------------------------------
/* This is an example sketch on how to find correct calibration factor for your HX711:
   - Power up the scale and open Arduino serial terminal
   - After stabelizing and tare is complete, put a known weight on the load cell
   - Observe values on serial terminal
   - Adjust the calibration factor until output value is same as your known weight:
      - Sending 'l' from the serial terminal decrease factor by 1.0
      - Sending 'L' from the serial terminal decrease factor by 10.0
      - Sending 'h' from the serial terminal increase factor by 1.0
      - Sending 'H' from the serial terminal increase factor by 10.0
      - Sending 't' from the serial terminal call tare function
      - Sending 'T' from the serial terminal call big tare function
   - Observe and note the value of the new calibration factor
   - Use this new calibration factor in your sketch
*/

#include "FastRunningMedian.h"
#include <HampelFilter.h>
HampelFilter tareHampel = HampelFilter(0.00, 19, 0.02);
FastRunningMedian<uint32_t, 19, 0> tareMedian;
#include <HX711_ADC.h>
// I modified the settings in config.h as follows:
// SAMPLES 64
// IGN_HIGH_SAMPLE 1
// IGN_LOW_SAMPLE 0

#define scale_DATA 0
#define scale_SCLK 1
#define scale_PWDN 2
#define scale_SPEED 22

//HX711 constructor (dout pin, sck pin)
HX711_ADC LoadCell(scale_DATA, scale_SCLK);

long t;

void setup() {
  Serial.begin(115200);

  pinMode(scale_PWDN, OUTPUT);
  digitalWrite(scale_PWDN, HIGH);    // keep high to stay on
  pinMode(scale_SPEED, OUTPUT);
  digitalWrite(scale_SPEED, LOW);    // keep high for high speed

  delay(300);
  Serial.println("preparing...");
  LoadCell.begin();
  long stabilisingtime = 5000; // tare preciscion can be improved by adding a few seconds of stabilising time
  LoadCell.start(stabilisingtime);
  LoadCell.setCalFactor(25830.0f); // user set calibration factor (float)
  LoadCell.setTareOffset(8380000);
  Serial.println("Startup + tare is complete");
}

void loop() {
  //update() should be called at least as often as HX711 sample rate; >10Hz@10SPS, >80Hz@80SPS
  //longer delay in scetch will reduce effective sample rate (be carefull with delay() in loop)
  LoadCell.update();

  //get smoothed value from data set + current calibration factor
  if (millis() > t + 250) {
    float i = LoadCell.getData();
    float v = LoadCell.getCalFactor();
    Serial.print("Load_cell output val: ");
    Serial.print(i);
    Serial.print("      Load_cell calFactor: ");
    Serial.println(v);
    t = millis();
  }

  //receive from serial terminal
  if (Serial.available() > 0) {
    float i;
    char inByte = Serial.read();
    if (inByte == 'l') i = -1.0;
    else if (inByte == 'L') i = -10.0;
    else if (inByte == 'h') i = 1.0;
    else if (inByte == 'H') i = 10.0;
    else if (inByte == 'T') multiple_tare();
    else if (inByte == 't') LoadCell.tareNoDelay();
    if (i != 't') {
      float v = LoadCell.getCalFactor() + i;
      LoadCell.setCalFactor(v);
    }
  }

  //check if last tare operation is complete
  if (LoadCell.getTareStatus() == true) {
    Serial.print("Tare offset: ");
    Serial.println(LoadCell.getTareOffset());
  }
}

void multiple_tare() {
  Serial.println("wait, calculating... this will take long... countdown:");
  for (int i = 0; i < 20; i++) {
    Serial.println(20 - i);
    delay(250);
    LoadCell.tare();
    uint32_t value = LoadCell.getTareOffset();
    tareMedian.addValue(value);
    double value_f = double(value) / double(2500000);
    tareHampel.write(value_f);
  }
  Serial.print("Tare by runningMedian: ");
  Serial.println(tareMedian.getMedian());
  Serial.print("Tare by HampelFilter: ");
  Serial.println(2500000 * tareHampel.readMedian());
  Serial.print("Tare by both filters: ");
  double tare_offset = 0.5 * (tareMedian.getMedian() + 2500000 * tareHampel.readMedian());
  Serial.println(tare_offset);
  LoadCell.setTareOffset(tare_offset);
}
```