olkal/HX711_ADC #13: data smooth index check needs to be fixed

davetpet, 2018-05-22:
currently when doing an update(), the code writing to the data smooth index handling looks to see if it is buffer size  - 1 and then sets it to 0.   Unfortunately, if you create this with new, or you are putting it into stack memory that has already been used, the index initially can be outside the range 0 - buffersize -1. 

To fix this the safest thing would be before it writes to the buffer, the index should be checked to ensure it is >= 0 and < buffer size and if it is outside that range, set to 0,.

An alternative would be to initialize the index to 0 on construction, but this does leave the possibility of some other bug setting it out of range and the update() continuing to overwrite memory it doesn't own.

olkal, 2018-05-22:
Hi
Sure, I can include some code to check that the readIndex is within range as you suggest. 
But could you please explain in a little bit more detail how the variable readIndex can be "initially out of range"? (I'm not familiar with the "new" operator)

olkal, 2018-05-22:
Got it, thanks a lot for the explanation! 
I have actually had this problem before with local variables, so I usually initialize those. 

As a fix I think I will initialize the index in the class declaration. We could in addition to that also do a range check of the index inside the function as you first suggested, that should be bullet-proof. 

You also mention initializing of the data array, but is that necessary? The whole data array will be overwritten with fresh values quickly and unlike the index, old garbage should not make any serious problems?

olkal, 2018-06-16:
Implemented Initializing of the index counter in the class declaration: readIndex = 0

closed 2018-06-16 by olkal