olkal/HX711_ADC #14: Feature request: setTareOffset()

jpk73, 2018-06-09:
Hi! First of all, thanks for this wonderful library! I have a question: there is getTareOffset() but no setTareOffset(): that would remove the need to tare the scale which is useful for permanent scale projects. Will you add such a function in the future? Best regards, jpk73

olkal, 2018-06-10:
Thanks jpk!
I'm not sure how useful that is though. The tare offset needs to be "fresh" as variations in temperature and probably other factors as well, will couse drifting in the load cell output, it's very senitive. So setting the tare offset as a fixed value at startup (if that's what you had in mind) will not work well.

jpk73, 2018-06-12:
Exactly that would be useful for my needs as I want to see long term changes including influences of humidity and temperature. BTW the load cells I have don't drift much with temperature... If you put a plant on a scale and want to see how it developes over weeks or months and switch on the scale every few hours you don't want to tare it - this kind of projects in mind I think a possibility to bypass the startup tare and manually punch in an offset value would be great. Regards!

olkal, 2018-06-16:
Okay then
setTareOffset(long newoffset) implemented. As in getTareOffset(), this is raw data values, not divided by the Calibration Factor. I did no testing but it should work.

closed 2018-06-16 by olkal

jpk73, 2018-06-17:
Many thanks, great help for me!

nervalo88, 2018-06-23:
Thanks for it ! 

I was about to make the same request.
My aim is to implement "deep sleep" of an ESP8266 between two cycles in order to reduce battery consumption. 
Then I'll store the TareOffset to the EEPROM and re-use it at the next cycle.
Agreed with the statement that this will be used only for long-run measurements, with a lack of accuracy.

Thanks again for this wonderfull job !

jpk73, 2019-06-19:
Hi, in the .h file you have "tareTimeOut = (SAMPLES + IGN_HIGH_SAMPLE + IGN_HIGH_SAMPLE) * 150", is that correct: 2 times IGN_HIGH_SAMPLE? Regards!