olkal/HX711_ADC #16: When I found the calibration factor, my scale still floating to -0.1 and -0.0

AJP2511, 2018-08-09:
Hello. 
First of all, thanks for this amazing work.
I followed the steps to calibrate and read one single sensor but when I remove any item from my scale the value showed to me change between -0.1 and -0.0, do you know how to fix this in zero? 

Thanks again

olkal, 2018-08-09:
Hi
This is quite common, as the HX711 chip is measuring and amplifing very small voltage variations from the load cell it takes very little noise to cause fluctuations like this. I can't exactly tell you how to fix it, but a few things might help (most of it is mentioned other places in this respository, so maybe you have already tried it though...)

Hardware:
- keep wiring between HX711 and the load cell short and away from other electronics
- twist the wiring between HX711 and the load cell 
- the HX711 10sps mode is usually recommended, it's slower but less noisy than the 80sps mode 
- don't use a load cell with much higher load capacity then you really need

Library:
- you can try to increase the number of samples (in the HX711_ADC.h file, line 14), especially if you are using the HX711 in 80sps mode

AJP2511, 2018-08-09:
Thanks for your really fast reply!

when i look at my HX711 board i just see the pin15 of CI attach to GND, so i working in 10SPS and have no way to change it (that's ok).

i'll test cutting the cables between HX711 and load cell (more short, like you said) and twist cables (this part really make me confuse but i'll try)! 

if i have some feedback i'll report here!

AJP2511, 2018-08-09:
ok, i don't believe in this result, but twist the wires between HX711 and load cell make a HUGE difference! 

my last question, my display show -0.0g and shakes to 0.0g i just need do a logic to exclude te minus singnal from zero?

olkal, 2018-08-11:
Hi, glad to hear that it made a difference. 
Yes, the -0.0 is easily fixed by in your code. Something like this:
`if((value < 0) && (value > -0.2)) value = 0;`

Note that twisting the load cell supply (AVDD + GND ref data sheet) wires and the output (INNA + INPA ref data sheet) wires into 2 twisted pairs is probably even better, and the more correct way to do it, but I have not tested this.

closed 2018-08-14 by olkal

JM-DfT, 2018-08-14:
Why does a twisted pair improve measurement results: https://en.wikipedia.org/wiki/Twisted_pair
Particularly useful on high ohmic signals.