olkal/HX711_ADC #18: Issue with reading data from load cell

layaannj, 2018-08-14:
Hi,

Board  - Arduino uno 
IDE    -     Arduino 1.8.1
Code   -   HX711_ADC/examples/Calibrate/Calibrate.ino
ADC + Amplifier - HX711 Breakout board (Green color)
Sensor  - 1Kg load cell
Vcc - 3.3
Gnd - Gnd pin
data - pin 3
clock -pin 4

I am trying to read data from load cell.I only get the output as 0 sometimes -0.00 even if i put some load on the load cell.

output -
 Wait...
Startup + tare is complete
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 0.00

some one please help me?

Thank you

AJP2511, 2018-08-14:
Did you try to use 5vdc? I think load cells don't work with less than 5vdc (HX711 work, but the cell need 5vdc to excitation)

layaannj, 2018-08-14:
ya i also try with 5v.

JM-DfT, 2018-08-14:
A load cell is a passive device and also works at 3.3V (but with lower resolution). Did you check if there's clock and data at the HX711 pins? Some HX711 boards have a different order of pins.

olkal, 2018-08-14:
The tare completes and you seem to get a value from the HX711 (even if it's only 0,00). This should indicate that there is communication between the MCU and the HX711 and that the clock/data wiring is okay. Is it possible that there is someting wrong with the the load cell or the wiring between the load cell and the HX711?

layaannj, 2018-08-27:
Hai,
Sorry for the late response .The connection are correct i cross checked .Now its giving output like this 
Wait...
Startup + tare is complete
Load_cell output val: 0.00      Load_cell calFactor: 696.00
Load_cell output val: -0.00      Load_cell calFactor: 696.00
Load_cell output val: -0.01      Load_cell calFactor: 696.00
Load_cell output val: -0.05      Load_cell calFactor: 696.00
Load_cell output val: -0.08      Load_cell calFactor: 696.00
Load_cell output val: -0.11      Load_cell calFactor: 696.00
Load_cell output val: -0.13      Load_cell calFactor: 696.00
Load_cell output val: -0.14      Load_cell calFactor: 696.00
Load_cell output val: -0.15      Load_cell calFactor: 696.00
Load_cell output val: -0.17      Load_cell calFactor: 696.00
Load_cell output val: -0.21      Load_cell calFactor: 696.00
Load_cell output val: -0.25      Load_cell calFactor: 696.00
Load_cell output val: -0.31      Load_cell calFactor: 696.00
Load_cell output val: -0.38      Load_cell calFactor: 696.00
 
When i press hard the value will change .when i put 100g it wont show any change.

olkal, 2018-08-27:
Hi
The library has no units. You set the calFactor so that the output value maches your prefered unit (raw data from HX711 is divided on the calFactor value). 

I don't know how you have cross checked the wiring between the load cell and HX711, but note that the color coding on the load cell wires can be different from batch to batch. Anyway, I don't think this is a library issue, it's most likely hardware related.

layaannj, 2018-08-28:
Hi,
Thank you for your response.I will check the hardware

closed 2018-10-12 by olkal