olkal/HX711_ADC #19: more speed to find the right weight

AJP2511, 2018-11-02:
hello guys.

have anny change can i do to make the program more fast to find my weight? actually he takes 2 seconds to give me my right weight but for my aplication i need that a little bit more fast.

i tried some alterations on smoothData and not work (in hexadecimal values).

thanks for you time.
Allan Johnson

olkal, 2018-11-03:
I suppose you are using the HX711 in 10SPS (samples per second) mode. With default library configuration with data set of 16+2 samples it will teoratically take 1.8 seconds to stabilize the value. You can make it faster by changing number of samples in the data set (line no 14 in the HX711_ADC.h file.: 
`#define SAMPLES 			16` 
Valid number of SAMPLES are 4, 8, 16, 32 or 64, you probably have 16 now, so changing SAMPLES to 8 or 4 will speed things up. However, you will then also have less samples to average, and depending on your hardware the accuracy might suffer a bit.

AJP2511, 2018-11-04:
Hello Olkal.

first of all, thanks for awnser my question.

i made the changes what you proposed and i don't see any difference in time and precision of scale (if have difference is much subtle).

i'm tryed with notepad++ and notepad to change the value but i got the same with all.

any idea?

olkal, 2018-11-04:
Is it possible that you have duplicates of the file HX711_ADC.h and are editing a different file then the one being compiled? Try to set SAMPLES to 0, that should throw a compile error.

AJP2511, 2018-11-04:
amazing. i left just your library in my arduino folder and all working fine now! 

i really appreciate your attention (times run like 1sec to find my weight now)

thanks alot olkal! 
have a nice day.

closed 2018-11-04 by AJP2511