olkal/HX711_ADC #20: read data step by step (slowly change)

lllkjh, 2018-11-16:
When I read the value, the resonse data increase or decrease slowly in step of aroud 3~4 g and finally stablize in the item weight. Is that possible to get the data immediately or much faster? I use the Ardiuno UNO board for testing. Code as below.

#include <HX711_ADC.h>
#include <LiquidCrystal.h>

float Weight = 0;
LiquidCrystal lcd(8, 9, 4, 5, 6, 7);
//HX711 constructor (dout pin, sck pin)
HX711_ADC LoadCell(10, 13);

void setup() {
  Serial.begin(9600);
  lcd.begin(16, 2);

  lcd.print("Welcome!");
  LoadCell.begin();
  long stabilisingtime = 100; // tare preciscion can be improved by adding a few seconds of stabilising time
  LoadCell.start(stabilisingtime);
  LoadCell.setCalFactor(696.0); // user set calibration factor (float)
  delay(3000);
}

void loop() {
  lcd.clear();
  LoadCell.update();
  Weight = LoadCell.getData();
  lcd.print("Weight = ");
  lcd.print(Weight);        //the true weight
  lcd.print(" g");

  delay(1000);
}

JM-DfT, 2018-11-16:
Removing this delay may just reveal even more that the measurement simply needs settling time. IMHO the observed effect may just be the mechanical settling time of the loadcell.

olkal, 2018-11-16:
> Removing this delay may just reveal even more that the measurement simply needs settling time. IMHO the observed effect may just be the mechanical settling time of the loadcell

No, that is not the case. With 1000ms delay in the loop you get a lot of unnecessary lag and the system will spend most of the time waiting rather than working, killing the overall performance. Only one conversion will be performed each second and the settling time will be >18 seconds with the default library configuration. Don't use delay() in the loop.

medolas, 2018-11-22:
It needs settling time, because you get average data. Default 16 samples + ignoring max and min sample.
You can set it down to 4 samples, but it won't be much faster. If you want immediate data use some other library  (e.g. hx711.h ). You will get tons of inacurate readings and at the end you will have to write your own code to get average data

Jer73, 2018-12-06:
I noticed the same. When putting a weight on the scale it takes up to 2 seconds to settle

olkal, 2018-12-07:
> I noticed the same. When putting a weight on the scale it takes up to 2 seconds to settle

Yes, the theroretical settling time with default library settings is 1.8 seconds. This is the time required for the HX711 to perform 16+2 conversions at 10SPS.

closed 2018-12-25 by olkal

olkal, 2018-12-25:
Bottom line:
Time consuming code in the loop() will increase settling time. Examples of time consuming code can be the use of delay(), or i.e. refreshing an LCD every time in the loop().