olkal/HX711_ADC #24: How to disable moving average

Luigi94100, 2019-03-27:
Hi,
Is there any way to disable the smoothing ?
I need no settling time and also the 80 sps for my project because it is in real time and the measurement increases very quickly.

Regards,
Yves

olkal, 2019-03-27:
Hi, you can try the function `getSingleConversion()` 
If a new conversion is ready it should return the data as a float and start the next conversion. If no new conversion is ready yet it returns -1.

Luigi94100, 2019-03-27:
Thank you olav 
I will try this as soon as I can

Luigi94100, 2019-03-28:
Hi,
I wrote this code but the load value is strange and constant.
I would be very grateful if you could help me.

```C++
#include <HX711_ADC.h>
float calValue= -21460/9.81;
float force;

unsigned int t_loop = 0;
unsigned int t_depart = 0 ;
unsigned int t_fin = 0 ;

HX711_ADC LoadCell(2,3); // 2 -> dout   3 -> clk

void setup() {
  // put your setup code here, to run once:
Serial.begin(9600);
 LoadCell.tareNoDelay();
 LoadCell.setCalFactor(calValue); // set calibration value (float)
 
}

void loop() {
t_loop = t_fin-t_depart ;
      if (t_loop >= 12500 ) // micros  
      {
        t_depart = micros() ;
        LoadCell.update();
        force = LoadCell.getSingleConversion();//scale.get_units() returns a float
        Serial.println(force);
      }
t_fin = micros() ;
}
```
Regards,
Yves

olkal, 2019-03-28:
Hi Yves
The function `getSingleConversion()` is no good, it's a bit embarrassing, but i think it was implemented without actually testing it. There is also some issues with your code. The library was never intended for reading out single conversions (quite the opposite), but I will make a fix for you by allowing only 1 sample in the config.h.

Luigi94100, 2019-03-28:
That would be awesome !

olkal, 2019-03-28:
Okay, files has been updated. Copy from respository or wait for the new release to be available in Arduino library manager. 

You need to do the following:
In library file config.h change the defined values to: SAMPLES 1, IGN_HIGH_SAMPLE 0, IGN_LOW_SAMPLE 0
In your sketch, setup() is missing some lines, it should look something like this:
```
void setup() {
 Serial.begin(9600);
 LoadCell.begin();
 long stabilisingtime = 2000;
 LoadCell.start(stabilisingtime);
 LoadCell.tareNoDelay();
 LoadCell.setCalFactor(calValue); // set calibration value (float)
}
```
In the loop, replace `force = LoadCell.getSingleConversion();` with `force = LoadCell.getData();`

Please let me know if this solves the issue.

BTW; I noticed that you have a negative calibration factor, that is unusual. If the reason for this is that you otherwise get negative readings, you could also flip the load cell 180° or switch 2 of the wires between the HX711 and the load cell.

BTW 2; depending on what other code you will include in the loop and your general requirements, you might want to consider an interrupt on the DOUT pin.

Luigi94100, 2019-03-29:
Thank you very much
Unfortunately, I did everything you said and the the value I get is constanly 0.

Here is my code

```C++
#include <HX711_ADC.h>

float calValue= 21460/9.81;
float force;

unsigned int t_loop = 0;
unsigned int t_depart = 0 ;
unsigned int t_fin = 0 ;

HX711_ADC LoadCell(2,3); // 2 -> dout   3 -> clk

void setup() {
  // put your setup code here, to run once:
 Serial.begin(9600);
 LoadCell.begin();
 LoadCell.setCalFactor(calValue); // set calibration value (float)
 LoadCell.tareNoDelay();
 
 
}

void loop() {
t_loop = t_fin-t_depart ;
      if (t_loop >= 12500 ) // micros  
      {
        t_depart = micros() ;
        LoadCell.update();
        force = LoadCell.getData();//scale.get_units() returns a float
        Serial.println(force);
      }
t_fin = micros() ;
}
```

olkal, 2019-03-29:
Hi, your code works fine when I test it. 
If you get a repeated 0.00 value this indicates that you don't have comunication between the MCU and the HX711 module, typically wiring or pin designation issue.

Luigi94100, 2019-03-29:
It is strange because I get values with the library written by bodge without touching anything.

olkal, 2019-03-29:
Yes that's strange. What kind of microcontroller are you using?

Luigi94100, 2019-03-30:
I am using the arduino due based on the Atmel SAM3X8E ARM Cortex-M3

olkal, 2019-03-31:
Hi, I have tested your code now with an Arduino Due using the same pins (2,3) at 80SPS, and it's still working fine here. I'm afraid I'm out of ideas of why you can't get it working.

Luigi94100, 2019-04-01:
Hi,
Maybe the difference comes from the ADC.
I am using the SparkFun Load Cell Amplifier.
Are the sck pin and the clk pin the same ?

olkal, 2019-04-01:
I don't have the Sparkfun module to test with, but there should be no difference as the two signal lines are connected directly to the HX711 chip, same as the modules I use. "dout" = "dat" and "sck" = "clk"
[https://cdn.sparkfun.com/datasheets/Sensors/ForceFlex/SparkFun_HX711_Load_Cell_v11.pdf](url)

Luigi94100, 2019-04-01:
My bad !
In my previous code I used CLK=2 and DOUT=3 whereas on the new code, I put HX711_ADC LoadCell(2,3).
That is why it worked with the bodge library and not with this one.
It now take 5 micros instead of 87 mills to get the data.
Thank you very much and sorry for my mistake.

olkal, 2019-04-01:
No problem, good that you figured it out :)

closed 2019-04-01 by olkal