olkal/HX711_ADC #25: getSingleConversionRaw() not working

jpk73, 2019-04-04:
Hi! I am successfully using your lib with an ADS1231, but I can't get getSingleConversionRaw() to work, it seems it is missing in the cpp file...? Regards and Thanks!!!

jpk73, 2019-04-04:
Also getSingleConversion() is missing...

olkal, 2019-04-25:
Hi, sorry for the delayed response...
getSingleConversionRaw() and getSingleConversion() was not working as intended and was therefore removed. I just forgot to remove them in the .h file, I will fix that.
If you need single conversion you can use samples = 1 in the config file. 
Raw data conversion is not supported at the time, but you can try the code below.

```
//HX711 conversion, raw data

int doutPin = 2;
int sckPin = 3;

unsigned long conversionData;
unsigned long t;

unsigned long getConversionData() {
  unsigned long data = 0;
  byte dout;
  for (uint8_t i = 0; i < (24 + 1); i++) { //read 24 bit data + set gain and start next conversion
    delayMicroseconds(1);
    digitalWrite(sckPin, 1);
    delayMicroseconds(1);
    if (i < (24)) {
      dout = digitalRead(doutPin);
      data = data << 1;
      if (dout) {
        data++;
      }
    }
    digitalWrite(sckPin, 0);
  }
  data = data ^ 0x800000; // if out of range (min), change to 0
  return data;
}

void setup() {
  Serial.begin(9600);
  pinMode(doutPin, INPUT);
  pinMode(sckPin, OUTPUT);
}

void loop() {
  byte dout = digitalRead(doutPin);
  if (dout == 0) { //dout low, new conversion ready
    conversionData = getConversionData();
    Serial.println(conversionData);
  }
}
```

closed 2019-04-25 by olkal