olkal/HX711_ADC #28: Start up without tare

spanout, 2019-04-30:
I love this library, thank you for creating it. It worked well first time. My application will be to have an esp-01 on deep sleep  and to wake up once per day to take a reading. If I start the the sample code with a mass on the sensor, then it tares it as zero and removing it takes the value negative. Is it possible to start the loadcell without a tare, using my setCalFactor to calibrate the gauge and give me the actual mass on the sensor at start up? I guess I'm asking if tare can be disabled on initiating the sensor?

Thanks

olkal, 2019-04-30:
Hi! 
Sure, just remove the "LoadCell.start" function in the setup part of your scetch, then tare will not be performed. The powerUp() and powerDown() functions can be usefull to save power if the HX711 is connected all the time. 

Keep in mind that the data set will contain old data when you wake up the unit, so you need to fill the data set with fresh conversions if you want to read an updated value that is represents the current load.

spanout, 2019-04-30:
Perfect ..thank you

spanout, 2019-05-08:
Hi, sorry to come back to this, but I have been away and unable to check until today. Removing the call to Start() gives be bad readings but with start, the value is perfect.

My setup code is:

```
  LoadCell.begin();
  long stabilisingtime = 2000; 
  LoadCell.start(stabilisingtime);
  LoadCell.setCalFactor(104510.80);
  while (!LoadCell.update());
```

And my reading code is:
```
void loop() {
  

  LoadCell.update();

  if (millis() > t + 10000) {
    outval(LoadCell.getData());
    t = millis();
  }

  
  
}
```

I have the CalFactor, is it possible to get this to work from startup with a weight on the sensor and to give the correct value at that point. The reason I need this is if there was a power outage, I do not want the load sensor to tare to the weight that is on it. 

Any help is greatly appreciated

olkal, 2019-05-08:
Hi!
If you by "bad readings" mean that you get a very high value readout, that makes perfect sense as you don't substract the tare value anywhere else in your code. Is this the case? If you put a known mass on the load cell, does the value increase same as the mass?

spanout, 2019-05-08:
Ahh.... got it... taking away that'tare value' makes it work perfectly.

Thank you and sorry for the stupid question

closed 2019-05-08 by spanout

olkal, 2019-05-08:
You're welcome :)

sidikhanrei, 2020-04-14:
Hi, this is interesting.
Can you post the sample code how to use this lib without start()?