olkal/HX711_ADC #29: Struggling to get 80 sps

Luigi94100, 2019-05-17:
Hi Olav,

Is the HX711 really able to provide 80 sps ?
I soldered the two pins on the back of the board together.
I wrote a code with an interrupt on the DOUT pin and another code using micros() function.
I get the same value several times so it creates a curve with aliasing.
I also tried the code you wrote in the issue #25.
I get a curve without aliasing but with much less SPS.

Regards,
Yves

olkal, 2019-05-17:
Yes, it should. I just did a test now using interrupt on dout with 80SPS enabled. Each conversion was about 10870us, so almost 92SPS. On my module the high rate is enabled when the solder jumper is un-bridged, but it might be different on yours. You can check this by measuring the voltage on HX711 pin 15 (the RATE pin ref data sheet), at high rate it should be same as the supply voltage.

Luigi94100, 2019-05-20:
Thank you for your advice.
I will try this.

Luigi94100, 2019-05-20:
It worked thank you !
The bridge was by default on the board.
I had to cut the circuit.

closed 2019-05-20 by Luigi94100

olkal, 2019-05-20:
Great!

olkal, 2019-05-20:
BTW; If you are interrested in faster operation, it's possible to connect an external oscillator, i.e. a 20mhz x-tal to pin 13 and 14 on the HX711 with 20pf caps to gnd. This will incresase the sample rate to about 18SPS at low rate and 144SPS at high rate. To disable the internal oscillator pin 14 has to be disconnected from from VCC by lifting the pin of the pcb pad using a solder iron. A bit difficult, but possible with tweezers and magifier.
 
[image]

Luigi94100, 2019-05-21:
Thank you for the advice. I will keep it in mind.
Does it have drawbacks ? for example on accuracy ?

olkal, 2019-05-21:
I don't think so, at least there is no drawbacks mentioned in the data sheet.

Luigi94100, 2019-05-21:
great thank you !

Luigi94100, 2019-07-05:
Hi Olav,
I am planning on buying a crystal as you suggested.
The only frequence that I found that gets an accurate sampling and which is over my need is 22,1184 MHz.
22,1184 x1000 / 138,240 = 160 SPS

I found a quartz with this frequency and 20 pF but it's SMT type.
I found another (not SMT) but with 18 pF.

Is it a problem if you don't have an accurate sampling or if you have 18pF instead of 20pF ?
Also, is the capacitor integrated to the oscillator or do you have to connect one ?

Regards,
Yves

olkal, 2019-07-09:
Hi Yves!
The sample rate does not need to be a whole number ea second, a 20mhz crystal will give you a sample rate of about 144,67 SPS, and that will be fine, timing is not critical. Changing the sample rate with a crystal does not really affect the measurement accuracy, but the overall performance (settling time etc) will be better with a i.e. 20mhz crystal as you will have a more samples to work with within the same time. 
A 2-pin crystal does not have internal caps, you must connect caps from ea pin to ground, 18, 20 or 22 pF will usually work for a 20mhz crystal, it's not that critical. For a crystal with more than 2 pins, check the data sheet.

Olav

Luigi94100, 2019-07-10:
Thank you very much !
20mhz is enough for me.
There is one last thing I don't get. Where is the capacitor on your picture ?

olkal, 2019-07-10:
Hi, the two 0805 smd caps that I used is soldered 'end to end' in between the crystal pins. The white wire is to ground.

[image]

Luigi94100, 2019-07-10:
Ok I get it, thank you !

burgauss, 2022-07-05:
> Hi Olav,
>
> Is the HX711 really able to provide 80 sps ? I soldered the two pins on the back of the board together. I wrote a code with an interrupt on the DOUT pin and another code using micros() [...]

Hy Ives,

What code did you implement to get the 80SPS operation? I can see you mention the interrupt functions

Regards Juan

Stigmen, 2023-07-13:
Having more samples wouldn't necessary increase accuracy even with averaging. 
Has someone measured on the HX711? On the ADS1232 it seems that SNR is worse with 80SPS even with 8x Averaging (https://gitlab.com/jousis/espresso-scale/-/blob/master/Hardware/Documents/RF_Filter_Tests.xlsx)

There is a fixed noise with every readout and Signal to Noise can be better if you'd let the Signal accumulate for 1 sec rather than taking 8 readings within one second and then averaging.
(Averaging 8x with 80SPS would improve SNR with square root of averages, whereas at 10SPS signal just adds up over 1 sec)

olkal, 2023-07-14:
This is well known, and in the ReadMe file for this library I advise to use 10SPS unless very fast sampling rate is needed. I think the 10SPS setting is most likely just the 80SPS with internal 8x oversampling. 
But I don't think the point here was to improve accuracy but rather to get a fast as possible response time, perhaps for a game controller or something else that need a fast sampling rate more than accuracy.