olkal/HX711_ADC #30: Can we disable tare?

nayanashettyft, 2019-06-10:
I have a simple code to setup and read the data from the load cell. However when I start the scale with load on it, it starts at 0 rather than the initial weight of the object on it.
Is there a way to disable the tare() so it gets the actual weight on the loadcell even at the start point?

Code I have:
```
HX711_ADC LoadCell(LOADCELL_DOUT_PIN, LOADCELL_SCK_PIN);
void setup() {
  LoadCell.begin(); // start connection to HX711
  LoadCell.start(2000); // load cells gets 2000ms of time to stabilize
  LoadCell.setCalFactor(200.0); // calibration factor for load cell
}
void loop(){
  LoadCell.update(); // retrieves data from the load cell
  long reading = LoadCell.getData(); // get output value
  Serial.print("HX711 reading: ");
  Serial.println(reading);
}
```

olkal, 2019-06-11:
Hi
Yes, you can remove the "LoadCell.start" function in the setup part of your scetch, then tare will not be performed when you power up (#28).

closed 2019-06-13 by olkal

nayanashettyft, 2019-06-15:
Hi thanks for the quick response, I tried this out yesterday and this fixed the issue for me.