olkal/HX711_ADC #31: how does using 80Hz change the code?

GoncaloFelicio, 2019-08-08:
Hi, 

Before anything thanks for creating this library! it's really helpful, i've been using it in a personal project and it uses 4 LC's (each with it's owns HX711) and i'm using 80Hz as i need the faster response.

I was wondering what changes in the code, if anything, when using 80 Hz, since i've been testing the examples 'read1x' and 'read2x' and i didn't notice a big difference between using 80Hz or 10Hz.

Thank you very much!

olkal, 2019-08-09:
Hi, and thanks!
The code does not care at all whether you set the HX711 at low rate 10hz or high rate 80hz. 

At low rate the HX711 use 8x internal oversampling, so changing from high rate to low rate you should see 8x slower settling time and 8x (3 bit) better resolution.

The libray have functions for measuring the real settling time and samples pr second, see example file Testing.ino.

GoncaloFelicio, 2019-08-09:
Thank You for the quick reply!

GoncaloFelicio, 2019-08-09:
Hi, i came into another issue, since im using 4 lc's at the same time, sometimes the readings dont sync with the HX711 readings and a wrong number comes up. How do I make it so it's always synced properly? should i change the 'milis' time ?

olkal, 2019-08-09:
Yes. Or you can remove the millis interval entirely by polling the update() function, it will return a positive value only if there is new conversions available. Something like this (for ea load cell):

```  
  //LoadCell.update();

  //get smoothed value from data set
  if (LoadCell.update()) {  // will return true only if there is new data
    float i = LoadCell.getData();
    Serial.print("Load_cell output val: ");
    Serial.println(i);
    //t = millis();
  }

I'm not sure if the serial com can cope with all the date from 4x LC's at 80hz though. If so, you could try changing from 9600 to 115200 baud.

closed 2019-11-21 by olkal