olkal/HX711_ADC #33: Scale without tare

gpalmia, 2020-01-25:
Hi,
I'm building a scale, with load cells, HX711 as a signal amplifier and Esp 32 as a microcontroller.
I chose Esp 32 because with sleep mode I can save much energy and therefore feed the system with a photovoltaic panel and a small lead-acid battery (the scale is in a remote site).
The problem I would like to solve concerns the  tare of the scale: I have to use the system continuously "sleeping down" (and "wakeing up") the controller every hour ... As you can imagine the system performs the tare at each wakeup so that already at the second weighing the object to be weighed is included in the tare and the weight shown on the display is zero.
Has anyone found themselves having to solve a similar problem?
Thank you for the answer...

olkal, 2020-01-25:
Hi
I'm not familiar with how the ESP32 behaves during sleep/wake, but the library includes som functions for tare manipulation that could be useful. 
Maybe something like this:
```
long myTarevalue;
myTarevalue = LoadCell.getTareOffset(); //store
//go to sleep...
//wake up...
LoadCell.setTareOffset(myTarevalue); //restore
```

aaromalila, 2020-01-25:
Hi,
Thank you for the great library! With my latest CoG-scale I encountered with a problem; everything else works like earlier but there is a linear drift in readings, depending on weight area it is from 0,1 g / 5 sec to maybe 0,2 g / sec. The same issue applies on both load cell-HX711 -pairs, so I think there is no HW problem. Have you bumped in to this kind of problem? If library's smoothing function is disabled the drifting turns slower but still exists. I tried also different sample sizes (config.h) and disabling the outlier -compensation.

DVB-Arduino, 2020-03-31:
So, on ESP8266 if I do not use LoadCell.start (stabilizingtime); at start, values ​​are added and go up to -370 kg. I also need to ensure that at start there is no tare and the weight is shown with a load, and not be reset to zero at the start of the system.

olkal, 2020-03-31:
If you want to be sure that tare is not executed at the start of the sketch, just leave out function LoadCell.start (stabilizingtime) in setup(). 

I don't quite understand what you mean by "values ​​are added and go up to -370 kg"? 
Note that the raw conversion value from HX711 is always a positive number, and the readout can only be negative if the tare offset is grater than the conversion value. So if you dont apply a tare offset I don't think it's possible to get a negative number readout.

DVB-Arduino, 2020-04-01:
negative is not a problem. Just changed the wires E +, E-. ))))
This is what I meant when I said values ​​are added:
Starting...
Startup + tare is complete
Load_cell output val: 0.00
Load_cell output val: 23.72
Load_cell output val: 47.44
Load_cell output val: 94.89
Load_cell output val: 118.62
Load_cell output val: 142.34
Load_cell output val: 166.06
Load_cell output val: 189.78
Load_cell output val: 213.51
Load_cell output val: 237.23
Load_cell output val: 260.96
Load_cell output val: 284.68
Load_cell output val: 332.13
Load_cell output val: 355.85
Load_cell output val: 379.57
Load_cell output val: 379.58
Load_cell output val: 379.58
Load_cell output val: 379.57
Load_cell output val: 379.57
Load_cell output val: 379.57
Load_cell output val: 379.57
Load_cell output val: 379.57

`//-------------------------------------------------------------------------------------
// HX711_ADC.h
// Arduino master library for HX711 24-Bit Analog-to-Digital Converter for Weigh Scales
// Olav Kallhovd sept2017
// Tested with      : HX711 asian module on channel A and YZC-133 3kg load cell
// Tested with MCU  : Arduino Nano, ESP8266
//-------------------------------------------------------------------------------------
// This is an example sketch on how to use this library
// Settling time (number of samples) and data filtering can be adjusted in the config.h file

#include <HX711_ADC.h>
#include <EEPROM.h>

//HX711 constructor (dout pin, sck pin):
HX711_ADC LoadCell(12, 14);

const int eepromAdress = 0;

long t;

void setup() {
  
  float calValue; // calibration value
  calValue = 22325.24; // uncomment this if you want to set this value in the sketch 
  //22325,24;
  #if defined(ESP8266) 
  //EEPROM.begin(512); // uncomment this if you use ESP8266 and want to fetch the value from eeprom
  #endif
  //EEPROM.get(eepromAdress, calValue); // uncomment this if you want to fetch the value from eeprom
  
  Serial.begin(9600); delay(10);
  Serial.println();
  Serial.println("Starting...");
  delay (5000);
  LoadCell.begin();
  
  long stabilisingtime = 2000; // tare preciscion can be improved by adding a few seconds of stabilising time
  //LoadCell.start(stabilisingtime);
 
  if(LoadCell.getTareTimeoutFlag()) {
    Serial.println("Tare timeout, check MCU>HX711 wiring and pin designations");
  }
  else {
    LoadCell.setCalFactor(calValue); // set calibration value (float)
    Serial.println("Startup + tare is complete");
  }
}

void loop() {
  //update() should be called at least as often as HX711 sample rate; >10Hz@10SPS, >80Hz@80SPS
  //use of delay in sketch will reduce effective sample rate (be carefull with use of delay() in the loop)
  LoadCell.update();

  //get smoothed value from data set
  if (millis() > t + 100) {
    float i = LoadCell.getData();
    Serial.print("Load_cell output val: ");
    Serial.println(i);
    t = millis();
  }

  //receive from serial terminal
  if (Serial.available() > 0) {
    float i;
    char inByte = Serial.read();
    if (inByte == 't') LoadCell.tareNoDelay();
  }

  //check if last tare operation is complete
  if (LoadCell.getTareStatus() == true) {
    Serial.println("Tare complete");
  }

}`

DVB-Arduino, 2020-04-01:
I took another new ESP8266, new HX711, and new 50 kg load cells. Connected on wires 5 cm to eliminate interference. The result is the same.
Starting...
Startup + tare is complete
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 22.24
Load_cell output val: 44.49
Load_cell output val: 66.73
Load_cell output val: 88.97
Load_cell output val: 111.22
Load_cell output val: 133.46
Load_cell output val: 155.71
Load_cell output val: 177.95
Load_cell output val: 200.19
Load_cell output val: 222.44
Load_cell output val: 266.92
Load_cell output val: 289.17
Load_cell output val: 311.41
Load_cell output val: 333.66
Load_cell output val: 355.90
Load_cell output val: 355.90
Load_cell output val: 355.90
Load_cell output val: 355.90
Load_cell output val: 355.90
Load_cell output val: 355.90
Load_cell output val: 355.90

olkal, 2020-04-01:
Hi
The function getData() returns an average of the library's rolling data set (default is 16+2 samples), but when you start, the  dataset is empty, thats why the first readout is 0 and then increasing until the dataset is filled up with data. 

If you need to have the data correct from the very first call to getData() in loop() you must first fill up the data set with conversions so that the average value reflects the actual data from HX711.
You could try to put this code in the setup() part of your sketch to fill up the data set before the loop() is started (this code has not tested, hope it works... ):

```
int s = LoadCell.getSamplesInUse() + 2; // get number of samples in data set
while( s > 0 ) {
  if (digitalRead(12) == LOW) { // HX711 dout pin is pulled low when a new conversion is ready
       LoadCell.getData(); // add data to the set and start next conversion
       s--;
   }
}
```

DVB-Arduino, 2020-04-01:
no, it did not work. I even increased the request period. Data is filled in for 3 requests of 500ms.
So the calibration went wrong too? Why do the values ​​eventually increase to 370 kg if there is nothing on the scale? I did the calibration according to the instructions: starting on an empty scale, set the weight to 90 kg, entered 90 kg into the port.

olkal, 2020-04-01:
I forgot to include the update() function in the code snippet, that's why it didn't work. Updated and tested version below, put it at the bottom of the setup() part of your sketch. 
```
///assuming that the dout pin is connected to D12:
  int s = LoadCell.getSamplesInUse() + 2; // get number of samples in data set
  while ( s > 0 ) {
    LoadCell.update();
    if (digitalRead(12) == LOW) { // HX711 dout pin is pulled low when a new conversion is ready
      LoadCell.getData(); // add data to the set and start next conversion
      s--;
    }
  }
```
The no-load value of 370 is probably because you didn't apply any tare offset. Similar; when I run your sketch (using my own calibration factor) on my 1kg loadcell without any mass, I get a no-load value of 12300gr. If I then send 't' in the terminal to perform tare offset, the no-load value is 0gr as it should be. If you don't want to use the library tare function, you will have to substract the zero value (370) in your sketch to get the actual mass. Or you can manipulate the library tare offset by using the functions described in post no. 2.

To check if calibration is correct, just add a known mass to the loadcell and observe if the value is increased by the correct number.

BTW; You probably know this, but the timing of 100ms (I assume this is what you changed to 500ms) in the example sketch does not affect the library data sampling at all, it's just the Serialprint interval.

DVB-Arduino, 2020-04-07:
Thanks. But this does not always work. WatchDog on the ESP8266 sometimes turns on.

olkal, 2020-04-07:
Try to include yield(); or delay(1); at the end of the above while() loop.

msevland, 2020-04-13:
Building a dual weight with individual HX711 sensors, but need a function for re-calibrating bouth sensors individually within a sketch.
When using the Calibration Sketch, I can get the raw number using getData(), but are not able to retreive this inside a sketch. Then I only get the calibrated number.. Tried getTareOffset(), but it's not the same as what I get using getData() in the calibration sketch.
How can I easily re-calibrate HX711 (not just setting a new setCalFactor()) using two individual sensors within the Read_2x_load_cell.ino sketch?

olkal, 2020-04-14:
@msevland please don't post new and unrelated issues within existing threads! I have re-posted your question under #37

closed 2020-04-14 by olkal