olkal/HX711_ADC #34: Linear drift in readings

olkal, 2020-01-25:
Hi,
Thank you for the great library! With my latest CoG-scale I encountered with a problem; everything else works like earlier but there is a linear drift in readings, depending on weight area it is from 0,1 g / 5 sec to maybe 0,2 g / sec. The same issue applies on both load cell-HX711 -pairs, so I think there is no HW problem. Have you bumped in to this kind of problem? If library's smoothing function is disabled the drifting turns slower but still exists. I tried also different sample sizes (config.h) and disabling the outlier -compensation.

_Originally posted by @aaromalila in https://github.com/olkal/HX711_ADC/issues/33#issuecomment-578431119_

olkal, 2020-01-25:
Hi
I have seen similar drifting, but from what I remember, not this much and always hardware related. Could you please test with the code below, it will just print the raw data from the HX711 to the serial console, no library functions. After some time calculate drift: **(last value - first value) / scale factor**

```
//HX711 conversion, raw data

int doutPin = 2;
int sckPin = 3;

unsigned long conversionData;
unsigned long t;

unsigned long getConversionData() {
  unsigned long data = 0;
  byte dout;
  for (uint8_t i = 0; i < (24 + 1); i++) { //read 24 bit data + set gain and start next conversion
    delayMicroseconds(1);
    digitalWrite(sckPin, 1);
    delayMicroseconds(1);
    if (i < (24)) {
      dout = digitalRead(doutPin);
      data = data << 1;
      if (dout) {
        data++;
      }
    }
    digitalWrite(sckPin, 0);
  }
  data = data ^ 0x800000; // if out of range (min), change to 0
  return data;
}

void setup() {
  Serial.begin(9600);
  pinMode(doutPin, INPUT);
  pinMode(sckPin, OUTPUT);
}

void loop() {
  byte dout = digitalRead(doutPin);
  if (dout == 0) { //dout low, new conversion ready
    conversionData = getConversionData();
    Serial.print("Value:");
    Serial.println(conversionData);
  }
}
```

aaromalila, 2020-01-25:
Hi, thank you. I used your test code but plotted the data. While doing that I connected the (integrated) usb-charger to the LIPO providing the power on and off and noticed immediately that there is a correlation to power source's voltage. This explains the same "problem" on both ADC channels. I'll check my connections tomorrow, maybe the PRO MINI's 3.3V is not stable.

[image]

olkal, 2020-01-25:
Yes, I think you're on to something there. Any variations in the HX711 supply voltage will also affect the measured voltage from the loadcell.  Nice graph.

aaromalila, 2020-01-26:
Solved. Stabilising the 3.3V with external regulator made the drift very small but in the end changing both amplifiers fixed the problem completely. So, HW-issue with low quality HX711s.

olkal, 2020-01-26:
Great!

closed 2020-01-26 by olkal