olkal/HX711_ADC #35: SCK_DELAY 1 necessary for ESP32

christian172, 2020-02-19:
more a FYI.
It seems that ESP32 needs the 1s sck_delay. Maybe you could add this to Readme

Previously i used Version 1.0.2 or 1.1.0 and updated now to 1.1.6 . With the new Version and no delay the values went  like this: From 0 to around 50 fine , and after that than suddenly jumped to around -33500 and stayed there in the 5 digits, slightly decreasing with increasing force.

With the delay everything seems to be back to normal.

Blubbaa, 2020-04-05:
Thank you, this fixed the exact same issue for me too. +1 for the idea to mention this configuration for the ESP32 in the README.

olkal, 2020-04-08:
Thanks! ESP32 info now included in readme.md and config.h

closed 2020-04-08 by olkal