olkal/HX711_ADC #36: How to get tare?

sidikhanrei, 2020-04-14:
This code work perfect, calibrating i so easy.

After I calibrated the calFactor using examples/Calibration.ino, its sabe to eeproom.

And i just run another examples and get calFactor from eeprom.
It's work perfectly.

But, when I restart the serial monitor with mass at the load cell, the value is reset to 0. And mass become negative when taking out mass from the loadcell.

How to start the code if we have mass at the loadcell without reset the value?

Thank you

_sorry, i m not using english_

olkal, 2020-04-14:
Hi, thanks!
If you need to restart the micro controller without loosing the old value I suggest that you save the tareoffset to eeprom and then restore it after restart: 
```
//before restart:
long tare_offset = LoadCell.getTareOffset(); //save this value to eeprom

//after restart, restore the previous value from eeprom and set:
LoadCell.setTareOffset(tare_offset);
```

sidikhanrei, 2020-04-14:
Hi Thank you,
I will try it tomorrow. 

Thank you for your fast respond.

sidikhanrei, 2020-04-15:
Hi, another questions is, how to start without LoadCell.start() and directly calculating object mass when serial monitor is opened?

olkal, 2020-04-15:
> Hi, another questions is, how to start without LoadCell.start() and directly calculating object mass when serial monitor is opened?

Try to put the code below in the setup() part of your sketch so that it runs before the loop is startet. That should fill up the data set. I will make a library function for this in the next release.

```
///assuming that the dout pin is connected to D4:
  int s = LoadCell.getSamplesInUse() + 2; // get number of samples in data set
  while ( s > 0 ) {
    LoadCell.update();
    yield();
    if (digitalRead(4) == LOW) { // HX711 dout pin is pulled low when a new conversion is ready
      LoadCell.getData(); // add data to the set and start next conversion
      s--;
    }
  }
```

olkal, 2020-04-19:
In latest release 1.2.0 there  are new start() functions with selectable tare/no tare. There are also new functions for refreshing the dataset before readiing in loop, after i.e. restart/reset

closed 2020-04-19 by olkal