olkal/HX711_ADC #38: Drift Issue using 50 MT x 4 LoadCell (200 MT)

sidikhanrei, 2020-04-20:
Hi, 
I have big LoadCell with total 200 MT.
Using this library on 5KG and 10 KG has no issue,
But when connected to 200MT LoadCell,
Reads of get data is unstable and fluctuated.

Do you have any idea to resolve this problem?

Thank you

olkal, 2020-04-21:
No I don't have any idea.
Could you provide a little more details like technical data for the load cell and amplitude of the fluctations.

sidikhanrei, 2020-04-21:
Here is detail of the spesifications

1. Type = ZSFL-A
2. Plc = 0.7
3. Sensitivity = 1.9993 mv/V
4. Zero Output = 0.60% FS
5. Input Resistance = 696.8 Ohm
6. Output Resistance = 701.5 Ohm
7. Safe Load Limit = 150% F.S
8. Insulation Resistance = <= 5000 Micro Ohm
9. Maximum Excitation = 20 V DC

or you can view at here

sidikhanrei, 2020-04-21:
For more detail, the drift is about -40 KG to 60 KG from the original scale, and it's fluctuated wildingly.

Do I need to modify your library for my own situations?

olkal, 2020-04-21:
The load cell data seems okay to me, don't see any reason there. However, the fluctations of -40kg to 60kg is at least 10 times as much as I would expect based on my (limited) experience on smaller load cells.
Have you set the HX711 rate to 10SPS (pin 15 low)?  
Could there be any electrical interference (nearby wires etc) that could have an effect on the signal/voltage between Load cell and HX711?

You could reduce noise by incrasing the value of SAMPLES in config.h, but I consider that to be last resort, if something is wrong it's better to find the root couse.

sidikhanrei, 2020-04-21:
I use getSPS() return with 10.93.
I dont know pin 15 Low mean, can you explain?

Oh one more, LoadCell is 5 wired (wire 5 is Shield), i just skip this shield or need to connect to hx711?

I will try increasing samples.
Hope its solve.

olkal, 2020-04-21:
If getSPS() return with 10.93 you have the 10SPS setting, thats good. 
Usually the shielding should be connected to the HX711 ground plane. Try it and see if there is any improvement.

olkal, 2020-04-21:
The HX711 power supply also matters. You could try replacing it with som something else.

sidikhanrei, 2020-04-21:
> The HX711 power supply also matters. You could try replacing it with som something else.

Any reference how to do that?

olkal, 2020-04-21:
I.e. if you are using a switched power supply now, try replaing it with a battery.

sidikhanrei, 2020-04-21:
what the best baudRate for this library? I use 9600, but the examples use 57600

olkal, 2020-04-21:
The baudrate is not important,, both 9600 and 57600 is fine.

sidikhanrei, 2020-04-21:
> I.e. if you are using a switched power supply now, try replacing it with a battery.

Its directly supply power to load cell or just hx711?

sidikhanrei, 2020-04-21:
> I.e. if you are using a switched power supply now, try replacing it with a battery.

Its directly supply power to load cell or just hx711?

> If getSPS() return with 10.93 you have the 10SPS setting, thats good.
> Usually the shielding should be connected to the HX711 ground plane. Try it and see if there is any improvement.

How I can place the shield from load cell to hx711. The board just has E+, E-, A+, A-, B+, B-

Where to place the shield wire of loadcell?

olkal, 2020-04-21:
The power supply you use for the HX711 and the micro controller. Voltage ripple from a (bad) switched power supply could disturb the adc conversions. If you are now using the same power supply that you used when testing the 10kg and 20kg load cells, it's probably fine and the problem something else.

The shielding from the load cell wire should be connected to ground on the HX711 module.

sidikhanrei, 2020-04-22:
Today I'm trying using SAMPLES 32 and 64, it's more stable. But it's make zero position is not 0 (in randomly time), so I need to refresh the tare using tareNoDelay() when it's not zero.

Any solution, or it is normal?

olkal, 2020-04-22:
No, it's not normal and you should not have to use 32 or 64 samples just to get it reasonable stable. 
So I think you have some kind of hardware problem, but I'm not sure what it is and I'm running out of ideas...

sidikhanrei, 2020-04-23:
I have changed HX711 with another one, and it's look same.
I've ordered new HX711 with Shielding. I'll try when its ready.

olkal, 2020-04-23:
Okay

olkal, 2020-09-02:
Closed, due to lack of activity.

closed 2020-09-02 by olkal