olkal/HX711_ADC #40: Overflow of data when there is a small offset at the beginning

Rabter1, 2020-06-07:
I dont know if this is the expexted behavior but i got a lot of issued because of this. 

When you connect a weight cell to the hx711 without a load. We get a value close to zero out of `conversion24bit()`
If you now pull on the weightcell, the data overflows and we get a value close to 2^24.

The problem is, that some weightcells and or HX711 Chips have an offset at the beginning... 
This means when you have no load on the weightcell than we get a value close to 2^24 out of `conversion24bit()`.
When you now start to put a weight on the weightcell then the data out of `conversion24bit()` overflows and it jumps close to zero. 
This means you meassure a huge negative value and you cant really meassure anymore...

**My Solution:**
By inverting the MSB we can shift the data we get out of `conversion24bit()` to the middle of the range. 
This results in a decrease in number when we pull on the weightcell and an increase when we put a weight on it. An offset at the beginning has no effect anymore because there is no overflow near to zero/no weight. 

I added this line to `conversion24bit()` which fixes the problem for me...
`data = data ^ 0b10000000000000000000000; //Invert MSB` 

Here i try to visualize it... xD
Before the "fix"
 0
|-------------------------|

After the fix
                  0
|-------------------------|

0 Is the position with no weight on the loadcell and |--| the range of the 24 bit.

olkal, 2020-06-07:
Hi
I'm a bit confused;
The HX711 output range is from min. 800000h to max. 7FFFFFh so this value is rolling over, and suppose to.
However, in `conversion24bit()` line 309 this range is converted to min. 0h to max. FFFFFFh by shifting the 24th bit, so the `conversion24bit()` output will not roll over. 

So a balanced wheat bridge load cell (0mV differencial) should be approx. in the middle of the `conversion24bit()` output range already, and if output is close to zero with no load this suggests that you are very close to the HX711 min. range of 800000h. But how it can then jump to max range with a little load, I don't understand.

Any offset in the load cell should not cause the value to overflow because if the HX711 is out of range, the HX711 output is saturated at min. 800000h or max. 7FFFFFh (according to the data sheet).

Are you sure you're not mixing up the HX711 output with the `conversion24bit()` output?

With this you are shifting the 23rd bit, is that a typo or intentional?
`data = data ^ 0b10000000000000000000000; //Invert MSB`

Rabter1, 2020-06-07:
Ohhh
Somehow i was using Version 1.1.6
And line 309 `data = data ^ 0x800000;` is missing in my version. 
It is basically the exact same thing i did aswell. 
So the (issue) was already fixed.... 

Sorry about that... I dont know why my version was not up to date.

closed 2020-06-07 by olkal

olkal, 2020-06-07:
Actually the code line `data = data ^ 0x800000; ` has been there since the beginning in version 1.0.0.

Rabter1, 2020-06-08:
I tried to replicate my error and I really cant understand what happened there because as you mentioned... the actual code line which fixes the overflow was already there...

The thing i cant understand though is:
I am working on a project with 2 collegues.
And everybody installed this library independently... 
We all had the same (wrong) code.. and everybody had the bug...
We used PlatformIO - Date was approximately Dec 2019

I try to gather the original code (we got by PIO) by my collegues (I dont have it anymore) 
I will report if I understand what happened there...

Out of a logic perspective this does not make any sense and should have never happened.

But i remember some other code which proves the mentioned code of line wasnt there:

It was something like:
```
if (data == 0x80000) {
 data = 0; 
}
```
But if there would have been that mentioned line of code this would have been redundant.

and then later we have this code:
```
if (data > 0) {
convRslt++;
...
}
```

Rabter1, 2020-06-15:
Okay i had this weekend a look into the issue. 
First of all: **The issue is not fixed yet!**
Second: My memory was wrong about the code! But i found the issue now...

There is still an overflow in the output of `conversion24bit()` 

**Why is this happening?:**
The HX-Chip sends out data from 0-0x8000 which is 0 - 24 bit..
Now... the datasheet tells us, that the HX-Chip sends 0x800000 when data is out of range MIN and 0x7FFFFF when out of range MAX

This means, when not out of range the highest data, that gets transmittet by the HX-module is **23bit long!!!**

But we invert the 24th bit which is wrong / doesnt change anything... 

**What we need to do:**
Invert the 23th bit instead of the 24th bit to arrange the output of the hx-chip so we dont get an overflow...
Change this line:
`data = data ^ 0x800000; // if the 24th bit is '1', change 24th bit to 0`
to this line:
`data = data ^ 0x400000; //Invert 23th bit`
fixes the issue!

BTW: XOR does not change the bit if its 1... it inverts the bit. And this is also what needs to happen:
if its 0 --> 1 
if its 1 --> 0

I can get more in detail if you want. I debugged this and i am pretty shure that this is the right fix. And i am 100% sure now that there is a bug right now.

olkal, 2020-06-15:
Hi!
I think that you might misunderstand the (somewhat confusing) relation between HX711 range and its output values; the HX711 outputs the lower half of its range is from 0x800000 to 0xFFFFFF (24th bit is 1) and then roll over and output the upper half of its range from 0x0 to 0x7FFFFF (24th bit is 0). Thats why we need to invert the 24th bit.

If you look at the reference driver in the manufacturers data sheet you will see that they do the very same thing; inverting the 24th bit.  

However, you are correct in that XOR is inverting the bit regardless if its 1 or 0, my comment in the code is somewhat misleading.

Rabter1, 2020-06-15:
I understand that the MSB is used to indicate if it is "negative" or "positive".
And because we are using unsigned we need to invert the MSB.. 
What is awkward again...  if i look closely at the datasheet as you mentioned i would also suspect, that i need to invert the 24th bit. 
And as you mentioned... in the reference driver they do indeed invertert the 24th bit... 

But if i debug the HX711 modules i have lying around... it seems that they never use the 24th bit and instead use the 23th bit to indicate positive or negative... the 24th bit is always 0 (even when negative)

I ask myself if my HX711 modules are somewhat different or faulty. 
Can you confirm, that your output out of `conversion24bit()` is continuous when switching between positive(pushing) and negative(pulling)?

Because i can provide my logs which show two things that indicate the conversion is not working for me:
1) With no load i get a value data = close to zero... instead it should be close to the middle of the range --> 24th bit 1 the rest 0 --> 0x800000 --> 83888608(int)
2) I get an overflow when switching between pulling and pushing

Rabter1, 2020-06-15:
### I found the issue:
This time for real...

I was looking in the fact, that the datasheet says we have 24 bits but i only got 23 bits... 

It turns out, that **you should read the Readme** and the comments in the config...
After i switched to ESP32 i didnt activate the delay between SCK Pin and Data-read..

Somehow the communication worked perfectly so i didnt realize the issue... 
But somehow the 24th bit was consistently wrong (zero)

**Adding the SCK_DELAY in config fixes the issue with ESP32**

Because the 24th bit was always 0 it didnt work... and it explains all of the behaviour i encountered... 

A possibility to improve this, would be to use interrupts to read the pin i guess...

olkal, 2020-06-15:
Ah, yes of course... glad to hear that you solved it :) Maybe we should make the sck delay default on.
Using interrupt on the dout to trigger when a coversion is ready (as shown in example) can be very useful, yes. An interrupt routine for actually reading out the 24bit value not so much.

Rabter1, 2020-06-15:
I digged a little bit deaper.
Here are some informations for people who are curious:
What exactly happened:
Obviously the HX registeres the short HIGH and LOW at the SCK pin otherwise it wouldnt have output any data...
Because ESP32 is reading very fast after setting the SCK pin the DOUT-Pin is still at the "old" value or in a switching state...
What basically happened was, that i was always reading the "old" value that should have been read before... this means, that i shiftet all bits by one.. 
This explains why my MSB was bit 23 and not 24
bit 23 was 22 and so on.. 
I lost the LSB 
Now i got double the resolution and double the gain as before because my bits are correct now...
I noticed it because i have also double the noise now. 

I dont know how close it is for the HX711-IC to register the short LOW duration of SCK pin but if you would want to optimize the speed of reading the 24bit i think you could drop that one Delay between `SCK LOW` and `SCK HIGH ` on ESP32
Then you would have only the delay between SCK HIGH and reading the data...

olkal, 2020-06-15:
Yes, its quite possible that the first delay can be omitted.  I never had the ESP32 connected to HX711 so I have never optimised it, but I will do that later some time. If you want you could try to comment out the delay at line 293 in the cpp file and see if it still works.

Rabter1, 2020-06-16:
For me it works to leave the delay of the sck LOW out... 
but i debugged it with an oscilloscope. And it shows, that the SCK low time is <0.2uS
The datasheet tells us, that we need at least 0.2uS low time at the SCK pin so i would not recommend to leave the delay out on ESP32. 

Nice to know: Without any delay SCK pin is arround 2.5MHz on ESP32. 

But there could be a potential issue:
I think ESP32 can shift other tasks in between your code when calling delay... so i think other code is executed inbetween our data communication.. At least it seems like it in oszilloscope.  This could result in a SCK high time > 50uS which is forbidden according to the datasheet. 

Thats why interrupts (using timer) might be not a bad idea at all...

olkal, 2020-06-16:
The ESP32 is a fast little bugger...
I don't think that timer interrupt for reading out the data will work well or be worth the effort, other interrupt routines can still run in between the timer interrupts and make problems. If you suspect that the sck high could be >50us due to other operations (wifi etc), you should enable SCK_DISABLE_INTERRUPTS. For maximum efficiency it might be possible to create a custom high resolution delay function, but I will probably not implement any such ESP32 specific code as long as the Arduino delaymicroseconds function works.