olkal/HX711_ADC #41: Flushing data after mcu performs a blocking task

mfxuus, 2020-06-24:
First of all, thanks for this awesome library!

I'm not sure if this is the case, but it feels like if I perform some task in  `loop()` that takes some time to finish (but only needs to be done once in a while), `LoadCell.getData();` takes some time to flush out data recorded in the past, and only after some time does it go back to fresh / current weight reading. I want to first confirm that this is indeed the case - since I could be imagining things I guess... But if this is the case, is there a way to clear the "stale" data so that `LoadCell.getData();` starts reporting more "real-time" data even after a blocking task on mcu?

Thanks!

olkal, 2020-06-24:
Hi!
Yes, that is correct; if you don't call `getData()` for some time the dataset will be "outdated". There is no way (yet) to get real-time data without refilling the whole dataset. If you have time to wait for that, you can call `resetSamplesIndex()` and then poll `getDataSetStatus()` until it returnes `true` (confirming that the dataset has been refreshed). 
However; is there any reason that you're not calling `getData()` from an interrupt routine rather then from the `loop()`?

mfxuus, 2020-06-25:
Hi! Thanks for the prompt reply!

Regarding interrupt routines - I guess it's because I'm quite new with all this and have just been following examples and templates. I've did a quick read-up on interrupt routines and it seems like I can use it to store the latest reading data into a global variable, then in my loop, where I had been reading from `LoadCell.getData()`, I should be able to read from the global variable?

However, I do have time to wait for the dataset to be reset and refilled, as I really only need the scale readings at a fairly low frequency - under normal circumstances, I only actually do anything with the readings every half sec or so, and if I get blocked doing some other tasks, it's no big deal for me to wait a few seconds for the readings to stabilize again. Would the interrupt be inefficient for my use case? In the sense that, I don't really need the MCU to be interrupted to read the scale data all the time - but when I do get a reading, I'm hoping the reading is "real-time" value. Pardon me if that is not how things work.

olkal, 2020-06-25:
Well, close enough... using interrupt on the HX711 DOUT pin, the microcontroller will read any new available conversions from HX711 immediately regardless of other ongoing tasks, so the dataset will always represent the latest values. This is usually more efficient, as you will call `getData()` exactly and only when needed. Check out the example file to see how its done. 

If you have time to wait, the `resetSamplesIndex()` and `getDataSetStatus()` method as decribed above should work, but I would prefer going the interrupt route myself.

closed 2020-06-25 by mfxuus