olkal/HX711_ADC #44: Getting Load_cell output val: nan

jalakamkiran, 2020-09-17:
[image]
 I am using a HX711 amplifier for my load cell..After calibration I am getting value as nan...what does that mean? I am pretty sure that my connections are correct.
Thanks in advance

olkal, 2020-09-17:
Hi! nan means "not a number". 
Does this happen with one of the example sketches or your own code?
What kind of microcontroller are you using?

jalakamkiran, 2020-09-17:
Hey @olkal, Thanks for the reply...the problem was not with the code but with the arduino..some how the pins I was using were damaged and didn't give results on those pins..I changed the board and it started working fine..

closed 2020-09-17 by jalakamkiran

olkal, 2020-09-17:
Great!