olkal/HX711_ADC #46: Erroneous data after setSamplesInUse()

sonic74, 2020-09-28:
When doing

```
LoadCell.begin();
long stabilizingtime = 2000; // tare preciscion can be improved by adding a few seconds of stabilizing time
boolean _tare = true; //set this to false if you don't want tare to be performed in the next step
LoadCell.start(stabilizingtime, _tare);
LoadCell.setCalFactor(calibrationValue); // user set calibration value (float)
LoadCell_2.setSamplesInUse(1)
```

I get wrong data. Shouldn't

https://github.com/olkal/HX711_ADC/blob/dff0de8503fdc335e77aef6521881ae8d5b76ae1/src/HX711_ADC.h#L112

be initialized to 0?

olkal, 2020-09-28:
Hi!
Initializing `lastSmoothedData` in the .h file should not requierd as it is being assigned to a value before use (in line 258 of the .cpp file). 

I'm not able to replicate an error using `setSamplesInUse(1)`, output data is as expected. Could you please provide the complete sketch?

sonic74, 2020-09-29:
> Initializing `lastSmoothedData` in the .h file should not be required as it is being assigned to a value before use (in line 258 of the .cpp file).

Only when calling getData() before setSamplesInUse()

olkal, 2020-09-29:
You're right. I have revised the header file according to your comment, thanks.

closed 2020-09-29 by olkal

sonic74, 2020-10-02:
Unfortunately, the missing initialisation doesn't seem to have been the culprit. As a workaround I now added refreshDataSet() after setSamplesInUse()