olkal/HX711_ADC #47: Implement in a new sketch

jaivishnuks, 2020-10-05:
Hi I am a beginner in Arduino. I wanted to use the HX711_ADC library in my sketch. I want to set the calibration factor myself.  My sketch is meant for issuing an abort signal to a process when the mass measured by load cell crosses a certain value. 
1. Can u pls tell me which all functions are to be used from the library
2. Without the serial communication is it possible to provide calibration factor by the user as part of sketch, so that every time the Arduino restarts, we dont need to provide calibration factor. Is there any zero offset value included like as in y = mx + c  If so, how to set the zero offset value. 

Pardon my questions as I am only a beginner.

olkal, 2020-10-05:
1: Available funtions are listed in the HX711_ADC.h file, line number 53-84.
2, setting calibration: This is shown in the example files, i.e in file Read_1x_load_cell.ino see line number 41 and 55 `setCalFactor(calibrationValue)`
2: Current zero offset/tare offset value can be read out with function `getTareOffset()` and can be set with function `setTareOffset(long newoffset)`

jaivishnuks, 2020-10-10:
Thank you. I had two  basic doubts.
1.  Suppose we are setting calibration value in our sketch, then aren't we supposed to set the tare_offest value also in order to complete the form as y = mx+ c , otherwise, we may end up with wrong values. Is my line of thought correct?  Also, can you pls explain how tare_offset is calculated during calibration?

2. Can we measure two load cell readings using a single HX711 ADC break-out board

olkal, 2020-10-10:
Hi!
1: You can set the offset value as a fixed value in the sketch, but your readout will then most likely not be zero, because the zero point will change a bit with temperature etc. So, it's better to apply the tare_offset value at each power up by first measuring the output and then apply the tare_offest value accordingly. It may also be convenient to be able to re-set the offset from a pushbutton in case of drifting output (same way as with a consumer kitchen scale).
The tare_offset is calculated before calibration, not during.

Simplified, the output is calculated like this:
Read rawoutput
tare_offset = rawoutput
output = (rawoutput - tare_offset) / calibration value

2: The HX711 has two load cell channels (A & B), but you can only do conversions from one at the time. The A channel have a max gain of 128 and the B channel has a max gain of 32, so the B channel can provide the same accuracy and resolution. So, it's possible, but usually not recommended. Use one HX711 for each load cell for best performance.

jaivishnuks, 2020-10-15:
Hi. Thank you for your reply.  Is it possible to compare the values of two load cells (using separate hx711 modules) at the same time from a single Arduino. For example, I want to compare two load cell outputs M1 and M2, and issue a command if M1 exceeds M2.

olkal, 2020-10-15:
Yes, the library supports multiple HX711 modules on single Arduino. See example file Read_2x_load_cell.ino

olkal, 2020-10-19:
Closing, as I'm consider this issue solved.

closed 2020-10-19 by olkal