olkal/HX711_ADC #53: Loadcell to be global, but assign Pins later on.

euboty, 2020-11-07:
Is there a way to construct the HX711_ADC object without assingning pins right away? For my program it is needed to ask for the pins of the HX711 in a dialog and then construct the HX711. All while Loadcell beeing a global object. Like this:
************
HX711_ADC Loadcell();

void setup(){
    dialog();    //user sets pin1 and pin2 through serial dialog
    HX711_ADC Loadcell(pin1, pin2);
}
************

Thanks for anyone who can help!

foofel, 2020-11-29:
you could create a pointer and asign int later on.

`HX711_ADC *loadCell = nullptr;
...
loadCell = new HX711_ADC(pin1, pin2)`

euboty, 2020-11-29:
I tried it but it throws an error: 
> request for member 'begin' in 'LoadCell', which is of pointer type 'HX711_ADC*' (maybe you meant to use '->' ?)

Code to recreate:
```#include <HX711_ADC.h>

byte pin1 = 0;
byte pin2 = 0;

HX711_ADC *loadCell = nullptr;

void setup() {
  Serial.begin(57600);
  setPins();
  loadCell = new HX711_ADC(pin1, pin2);
  loadCell.begin();
}

void loop() {
  // other stuff for loop
}
void setPins(){
  /* 
   *  Here would be a function that let the user input what is pin1 and pin2.
   */
  pin1 = 4;     //just for Example
  pin2 = 5;     //just for Example
}```

This is not urgent at all I'm just curious how I can do it .

foofel, 2020-12-02:
The error mesage even show the solution :) If you use a pointer, you have to use `->` instedt of `.` to access the object, therefore you need to change `loadCell.begin();` to `loadCell->begin();`. you have to do this everyehwere you use the object.

euboty, 2020-12-07:
Thank you so much. I never worked with pointers before, but will sure do now.

closed 2020-12-07 by euboty