olkal/HX711_ADC #56: Reset weight value to zero

Polollo, 2020-12-15:
hi, i have an issue with my sketch.
when i'm measuring an item and i remove it,  the weight value  not return to 0 . i would like to know if it is possible to recall a void or a function that reset the weight value or return to tare value propertly, so i can assign this function to a phisic button.
like the tare button on a kitchen scale for example.

thanks for now...waiting news. regards

olkal, 2020-12-15:
Hi!
You can use one of the following library functions:
```
tare(); //zero the scale, wait for tare to finnish (blocking)
tareNoDelay();  //zero the scale, initiate the tare operation to run in the background (non-blocking)
```

Polollo, 2020-12-15:
Void loop (){
My sketch
My sketch
Ecc..
If(digitalRead(button) == HIGH) {LoadCell.tare()}
}

In this way? I have tried already but didn't work ... can you confirm that
is the correct way to recall?

Thanks a lot for helping me

olkal, 2020-12-15:
More or less correct yes, but perhaps this would be better:
```
void loop () {
   static boolean lastbuttonstate = LOW;
   boolean buttonstate = digitalRead(button);
   if (lastbuttonstate != buttonstate) {
      if (buttonstate == HIGH) {
          LoadCell.tareNoDelay();
      }
      lastbuttonstate = buttonstate;
   }
}
```

Also see the example file `Read_1x_load_cell.ino`, tare/zero function is already implemented by serial terminal input:
```
  // receive command from serial terminal, send 't' to initiate tare operation:
  if (Serial.available() > 0) {
    float i;
    char inByte = Serial.read();
    if (inByte == 't') LoadCell.tareNoDelay();
  }

  // check if last tare operation is complete:
  if (LoadCell.getTareStatus() == true) {
    Serial.println("Tare complete");
  }
```

Polollo, 2020-12-15:
Thanks a lot man . I will try this as soon as possible. If it works i would
like to send you a little donation via paypall to thank you for the help,
so if you want reply with the paypall email.

olkal, 2021-02-12:
I’m closing this issue because it has been inactive for some time. 
Please reopen if you still encounter this issue with the latest version.
Thank you!

closed 2021-02-12 by olkal