olkal/HX711_ADC #57: Stuck in endless while loop startmultiple() with _tare set to false

pvakharwala, 2021-01-02:
I am trying to get data from 2 load cells but the startmultiple function returns some arbitrary value when _tare is set to false how shall I combat that? Also the while loop runs endlessly if I use the 2x load cell example.
Please help

olkal, 2021-01-02:
Hi!

If you set tare false in startmultiple function, the tare offset avlue will not be set and the output value without load will not be zero (it will be some number very far from zero). If you don't wish to let the startmultiple function calculate and set the tare offset, you can do it later in the loop, see example file, "send 't' from serial terminal", line no 87 in the  2x load cell example.

The endless loop issue under discribed circumstances has now been fixed in 1.2.6, thanks.

pvakharwala, 2021-01-03:
How can I set it manually like apply that offset value. Basically what is happening is I calibrate my sensor using the calibration sketch then I load my code and when it runs there is some value other than zero as you described. So will that value be same everytime the sensors and board powers on? And how shall I get that value and store it and use it in my code?

pvakharwala, 2021-01-03:
> Hi!
>
> If you set tare false in startmultiple function, the tare offset avlue will not be set and the output value without load will not be zero (it will be some number very far from zero). If you [...]

How can I set it manually like apply that offset value. Basically what is happening is I calibrate my sensor using the calibration sketch then I load my code and when it runs there is some value other than zero as you described. So will that value be same everytime the sensors and board powers on? And how shall I get that value and store it and use it in my code?

olkal, 2021-01-03:
Yes the output value will be aprox. the same ea time you start, but be aware that it may change a bit by temperature etc.
If you want to manually apply the zero offset value you first need to obtain the value like this:
- start without any mass on the loadcell
- run the function `loadcell.tare()` or `loadcell.tareNoDelay()` to calculate the zero offset value
- run `long zero_offset_val = loadcell.getTareOffset()` and note the returned value `zero_offset_val`

You can now use this value to manually set the zero offset in your sketch: `loadcell.setTareOffset(zero_offset_val)`

olkal, 2021-02-12:
I’m closing this issue because it has been inactive for some time. 
Please reopen if you still encounter this issue with the latest version, or if the postings in this thread don't solve your problem .
Thank you!

closed 2021-02-12 by olkal