olkal/HX711_ADC #58: How to switch the AB channel

dingrenjun, 2021-01-16:
I need to use a HX711 to connect 2 sensors, how to switch the AB channel, I set 128/32, the channel cannot be switched

olkal, 2021-01-16:
Hi!
You can select the B-channel like this: `LoadCell.setGain(32);`

However; the library is not really intended for applications where you need to switch back and forth between channel A and B, but it may work if you also set the `calibrationValue` and tare/zero load offset every time you switch. Still, there is only one dataset and until that has been "flushed" with new readings from the selected channel, the average value will be a mix of channel A and B.
A better approach if you need to utilize both channels would probably be to use a seperate instance of the library for each cannel.

Is there a particular reason that you're not using one dedicated HX711 for each loadcell? Thats the best solution.

olkal, 2021-02-12:
I’m closing this issue because it has been inactive for some time.
Please reopen if you still encounter this issue with the latest version, or if the postings in this thread don't solve your problem .
Thank you!

closed 2021-02-12 by olkal

felix-schreib, 2021-02-23:
Hi,
i want to use the B channel, for some reason the loadcell only works until half of the specified range in the A channel. Even though I use the correct calibration value. 
Where in `setup()` do i need to call `LoadCell.setGain(32);` ?
Thanks

olkal, 2021-02-23:
Hi,
you can put `LoadCell.setGain(32);` anywhere after `LoadCell.begin();`. 
Or you can just replace `LoadCell.begin();` with `LoadCell.begin(32);`.

felix-schreib, 2021-02-23:
Its working great now, thanks you!