olkal/HX711_ADC #59: Persistent calibration

jakubsacha, 2021-01-20:
I'm using this piece of code to handle loadcell as a break pedal (racing games).

```

//-------------------------------------------------------------------------------------
// Thrustmaster T3PA brakeMod for loadcell
// Christopher Grimm 12.07.2018
//-------------------------------------------------------------------------------------

#include <HX711_ADC.h>
#include "Joystick.h"

HX711_ADC LoadCell(7, 8);

long t;

float min_read = 5;
float max_read = -1000;
unsigned long time_now = 0;
int calibration_period_in_s = 60*5;
float scaling_value = 255;
bool recalibration = false;

bool FLAG_debugging = false;

Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID, 
  JOYSTICK_TYPE_JOYSTICK, 0, 0,
  false, false, false, false, false, false,
  false, false, true, true, false);

// Set to true to test "Auto Send" mode or false to test "Manual Send" mode.
const bool testAutoSendMode = false;

const unsigned long gcCycleDelta = 1000;
const unsigned long gcButtonDelta = 500;
const unsigned long gcAnalogDelta = 25;
unsigned long gNextTime = 0;
unsigned int gCurrentStep = 0;

void setup() {
  Serial.begin(250000);
  LoadCell.begin();
  long stabilisingtime = 0; // tare preciscion can be improved by adding a few seconds of stabilising time
  LoadCell.start(stabilisingtime);
  LoadCell.setCalFactor(696.0); // user set calibration factor (float)

  Joystick.setAcceleratorRange(0, 260);
  Joystick.setBrakeRange(0, scaling_value);
  
  if (testAutoSendMode)
  {
    Joystick.begin();
  }
  else
  {
    Joystick.begin(false);
  }
  
  pinMode(A0, INPUT_PULLUP);
  pinMode(13, OUTPUT);
  
}

void loop() {

  // memory allocation
  float i;
  float ii;
  float bandwidth;
  
  // read value from loadcell
  LoadCell.update();
  i = LoadCell.getData();

  // read time
  time_now = millis();
  
  // adjust scaling factors during the first 10 seconds
  if (time_now < (calibration_period_in_s * 1000) )
  {
    if ( (time_now % 1000) > 500)
    {
      digitalWrite(LED_BUILTIN, HIGH); 
      }
      else
      {
        digitalWrite(LED_BUILTIN, LOW);
        }
    if ( i < min_read)
    {
      min_read = i;
      }
  
    if ( i > max_read)
    {
      max_read = i;
      }      
      
  }
  else
  {
    // do recalibration of scaling factors to ensure reaching the limits
    if (recalibration == true)
    {
      recalibration = false;
      bandwidth = max_read - min_read;
      min_read += 0.01 * bandwidth;
      max_read -= 0.01 * bandwidth;
      }
    
    digitalWrite(LED_BUILTIN, LOW);
    }

  // scale the value
  ii = (i - min_read) / ( max_read - min_read  ) * scaling_value;
  if (ii <= 0)
  {
    ii = 0;
    }
    if (ii >= scaling_value)
  {
    ii = scaling_value;
    }

  // for debugging: write to serial monitor
  if (FLAG_debugging == true)
  {
    Serial.print(  i );
    Serial.print("\t");
    Serial.print(  max_read );
    Serial.print("\t");              
    Serial.print( min_read );
    Serial.print("\t");              
    Serial.print( ii );          
    Serial.println();
  
    delay(20);
  }

  // write value to joystick api
  Joystick.setBrake( ii );

  if (testAutoSendMode == false)
  {
    Joystick.sendState();
  }
}
```
The code is coming from here: https://www.thingiverse.com/thing:3043797

Mechanism behind this code is to recalibrate hx711 after the bootup. That means that after each startup.

I made some alterations to the original code:
```
float min_read = 10000;
float max_read = -10000;
int calibration_period_in_s = 10;
bool recalibration = true;
```
to
```
float min_read = 5;
float max_read = -1000;
int calibration_period_in_s = 60*5;
bool recalibration = false;
```
hoping that it will make my break pedal consistent. 

Sadly it seems like each time i turn on the device it provides different min and max values. Would you have any advice to me how to calibrate the device once and reuse it between startups?

olkal, 2021-01-21:
Hi!
That the max value varies is obvious as it is a result of the applied human force after bootup, but the min value should be pretty consistent. If not, check for possible noise sources close to the load cell, the HX711 and wiring between them. Keep wiring between the load cell and the HX711 as short as possible.

It should be possible to use fixed min/max values for an application like this, I don't see why there should be any need to re-calibrate after each bootup. Also the deadband value implemented in line 103-109 should also give some room for drifting. 

Have you tried to replace the code in line 79-109 (re-calibration) with fixed values for min_read/max_read?

MBENKI, 2021-02-11:
Hi @olkal ! i have put together load cells for a scale that i use with Arduino, so i keep the calibration value in the EEPROM memory i use it daily, and i have to turn it off each time, when i turn it on i always have to remake the calibration, i would like to ask what can i do to only turn my scale on without remaking the calibration (i have planned a button to press whenever i decide to re-calibrate though). i took a look on the HX711_ADC.cpp file but i must confess i don't know how i should use those function or which i should use. 
Please i would appreciate it very much if you could help me :)
Thank you so much.

olkal, 2021-02-11:
Hello @MBENKI, 
Storing the value in EEPROM like shown in the examples should work, provided that you use av microcontroller with internal or emulated EEPROM. Re-calibration is normally not needed at all. Could you please post your sketch?

MBENKI, 2021-02-12:
Hi @olkal ok so i have the sketch in which i use two libraries i created (it is one of slave scales provided with one Arduino all of them comuncating with the master which is a Raspberry ) in this sketch i call the _calibrate()_ method in the library _calibration_mod.h_   here's the sketch followed by the .cpp file in the library : 

`#include <calibration_mod.h>
#include <printu8g64.h>
#include <HX711_ADC.h>
#include <EEPROM.h>
#include <Wire.h>
#include <SPI.h>

//pins:
const int buttonPin = 2; // the number of the pushbutton pin
const int HX711_dout = 4; //mcu > HX711 dout pin
const int HX711_sck = 5; //mcu > HX711 sck pin
const int eepromCalibAddr = 2;
const int eepromi2cAddr = 3;
const int precisionScale = 10; // round of weight precision
int buttonState = 0; // variable for reading the pushbutton status
int i2cAddr = EEPROM.get(eepromi2cAddr, i2cAddr);
unsigned long t = 0;
int currentWeight = 0;
//HX711 constructor:
HX711_ADC LoadCell(HX711_dout, HX711_sck);

void setup() {
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }
  Serial.println("Serial ok ...");
  Serial.println("ADDR: " + String(i2cAddr));
  EEPROM.put(eepromi2cAddr, i2cAddr);
  Serial.println("read ADDR after write: " + String(EEPROM.get(eepromi2cAddr, i2cAddr)));
  if (i2cAddr < 4 || i2cAddr > 119) {
    i2cAddr = 4;
  }
  Serial.println("ADDR after test: " + String(i2cAddr));
  Serial.println("read ADDR after test: " + String(EEPROM.get(eepromi2cAddr, i2cAddr)));
  Wire.begin(i2cAddr);
  Wire.onRequest(requestEvent); // register event
  Wire.onReceive(receiveEvent); // register event
  pinMode(buttonPin, INPUT);
  LoadCell.begin();
  unsigned long stabilizingtime = 3000; // preciscion right after power-up can be improved by adding a few seconds of stabilizing time
  boolean _tare = false; //set this to false if you don't want tare to be performed in the next step
  LoadCell.start(stabilizingtime, _tare);
  if (LoadCell.getTareTimeoutFlag() || LoadCell.getSignalTimeoutFlag()) {
    draw64("timeout!", 1500);
    while (1);
  }
  else {
    LoadCell.setCalFactor(1.0); // user set calibration value (float), initial value 1.0 may be used for this sketch
    draw64("startup ok", 1500);
  }
  while (!LoadCell.update());
  calibrate(); //start calibration procedure
}

void loop() {
  static boolean newDataReady = 0;
  const int serialPrintInterval = 3000; //increase value to slow down serial print activity
  Serial.println("read in EEPROM i2c / EEPROM calib: " + String(EEPROM.read(eepromi2cAddr)) + " / " + String(EEPROM.read(eepromCalibAddr)));
  executePressedButton();
  findDisplayWeight(newDataReady, serialPrintInterval);
  LoadCell.tareNoDelay(); //tare
  // check if last tare operation is complete
  if (LoadCell.getTareStatus() == true) {
    draw64("tarat. ok", 2000);
  }
}

[... 83 lines cut ...]
    float newCalibrationValue = LoadCell.getNewCalibration(known_mass); //get the new calibration value
    draw64("last val. EEPROM: ", 2000);
    draw64(String(EEPROM.get(eepromCalibAddr, calib)), 2000);
    EEPROM.put(eepromCalibAddr, newCalibrationValue);
    draw64("new val. calib.", 2500);
    draw64(String(newCalibrationValue), 2500);

    #if defined(ESP8266)|| defined(ESP32)
    EEPROM.begin(512);
    #endif
    EEPROM.write(eepromCalibAddr, newCalibrationValue);
    #if defined(ESP8266)|| defined(ESP32)
    EEPROM.commit();
    #endif
    
    draw64("calib. ok", 1500);
    draw64("*******", 1500);
  } else {
    LoadCell.refreshDataSet();
    draw64("calibrato :)", 1500);
    draw64("*******", 1500);
  }
}`

Thank you in advance @olkal :)

olkal, 2021-02-12:
Hi!
I didn't read it all but try this first:

`const int eepromCalibAddr = 2;
const int eepromi2cAddr = 3;`
As eepromCalibAddr in adr 2 is an int and take up two bytes it will be overwriten by eepromi2cAddr, change it to:
`const int eepromCalibAddr = 2;
const int eepromi2cAddr = 4;`

MBENKI, 2021-02-12:
Yes i have thought of it but it retrieves the right values stored in both EEPROM locations. the problem is even if i still got the good calibration value, the scale doen't give the right results unless i go through the tare again ... so i thought maybe instead of going through the tare just call Loadcell.refreshData() or something, with no concluding results unfortunately ...

olkal, 2021-02-12:
Are you sure that the value that you save to EEprom is the same that you retrieve from the EEprom after a reboot? It's difficult to see how it can be retrieved correctly:

`calib` will take 4 bytes of course, not 2 as I first wrote, so the value eepromi2cAddr must be at least 6. As it is, the values `int i2cAddr` and `float calib` are actually sharing two bytes on the EEprom.
And this line: `EEPROM.write(eepromCalibAddr, newCalibrationValue);` is wrong, and should be deleted. You can not "write" a float, you have to use "put". 

And further: `EEPROM.read(eepromCalibAddr)` in your Serial print code will actually only read the 1st of the 4 bytes in the float, you have to use something like `EEPROM.get(eepromCalibAddr, float value)`.

MBENKI, 2021-02-15:
Oh ok i have first replaced _EEPROM.read_ by _get_ and _EEPROM.write_ by _put_  and changing the _eepromCalibAddr_ to 10 and _eepromi2cAddr_ to 5, i thought to use the method setCalFactor() passing the previous calibration value as argument and i call calibrate() only if the calibration value from EEPROM is bigger than 22 or smaller than 21, but still i get strange values now my setup code looks like this : 

  ` const int buttonPin = 2; // the number of the pushbutton pin
    const int HX711_dout = 4; //mcu > HX711 dout pin
    const int HX711_sck = 5; //mcu > HX711 sck pin
    const int eepromCalibAddr = 10;
    const int eepromi2cAddr = 5;
    const int precisionScale = 10; // round of weight precision
    int buttonState = 0; // variable for reading the pushbutton status
    int i2cAddr = EEPROM.get(eepromi2cAddr, i2cAddr);
    float prevCalib = EEPROM.get(eepromCalibAddr, prevCalib);
    unsigned long t = 0;
    int currentWeight = 0;
    //HX711 constructor:
    HX711_ADC LoadCell(HX711_dout, HX711_sck);
    
    void setup() {
      Serial.begin(9600);
      while (!Serial) {
        ; // wait for serial port to connect. Needed for native USB port only
      }
      Serial.println("ADDR: " + String(i2cAddr) + " calib: " + String(prevCalib));
      EEPROM.put(eepromi2cAddr, i2cAddr);
      EEPROM.put(eepromCalibAddr, prevCalib);
      if (i2cAddr < 4 || i2cAddr > 119) {
        i2cAddr = 4;
      }
      Wire.begin(i2cAddr);
      Wire.onRequest(requestEvent); // register event
      Wire.onReceive(receiveEvent); // register event
      pinMode(buttonPin, INPUT);
      LoadCell.begin();
      unsigned long stabilizingtime = 3000; // preciscion right after power-up can be improved by adding a few seconds of stabilizing time
      boolean _tare = false; //set this to false if you don't want tare to be performed in the next step
      LoadCell.start(stabilizingtime, _tare);
      if (LoadCell.getTareTimeoutFlag() || LoadCell.getSignalTimeoutFlag()) {
        draw64("timeout!", 1500);
        while (1);
      }
      else {
        LoadCell.setCalFactor(prevCalib); // user set calibration value (float), initial value 1.0 may be used for this sketch
        draw64("startup ok", 1500);
      }
      while (!LoadCell.update());
      if (prevCalib < 21 || prevCalib > 22) {
        calibrate(); //start calibration procedure
      }  
    }
`

MBENKI, 2021-02-15:
@olkal it seems that having the right calibration value is not sufficient, something else is missing i can't figure out what ?? :(

olkal, 2021-02-15:
Very good! 👍

olkal, 2021-04-18:
I’m closing this issue because it has been inactive for some time.
Please reopen if you still encounter this issue with the latest version.
Thank you!

closed 2021-04-18 by olkal