olkal/HX711_ADC #60: Question: Hardware SPI possible/advisable?

n-gineer, 2021-01-20:
To speed up readings, is it possible to implement this library using hardware SPI? are there any other tradeoffs?

olkal, 2021-01-20:
I don't think so, as the HX711 serial interface is different from SPI. 
You could speed readings up by modifying the library to use direct port manipulation, but this will not be implemented in the officiel library version due to compability issues. A faster MCU would also be a posibilty, but there are limits for how fast you can go, e.g. with the ESP32 you need to enable the 1us clk delay in the config.h file.

That said, I just measured 145us to read out a conversion with an Arduino Due, so quite fast for most applications.

n-gineer, 2021-01-20:
Understood; thanks! 145 us is plenty fast for what I was planning to do.

closed 2021-01-20 by n-gineer