olkal/HX711_ADC #61: Scale working until reset

Edward-Sams, 2021-02-11:
Hello World.

I'm working on building a digital readout of how much beer is left in my home-brew.  I have the code working (currently testing with a 12oz can) and it is displaying the percentage left; however, if the power goes out, I am losing the initial 0 value of the scale.  I believe I need to write the 0 value to the eeprom, but I'm thinking I have my code in the wrong section or I have still have code that is zeroing it out.   Does anyone see what I'm doing wrong here?  -thanks

#include <HX711_ADC.h>
#include <EEPROM.h>
#include<LiquidCrystal.h>
LiquidCrystal lcd(12,11,5,4,3,2);

//pins:
const int HX711_dout = A1; //mcu > HX711 dout pin
const int HX711_sck = A5; //mcu > HX711 sck pin

//HX711 constructor:
HX711_ADC LoadCell(HX711_dout, HX711_sck);

const int calVal_eepromAdress; //=0;
unsigned long t = 0;

void setup() {
  Serial.begin(57600); 
  Serial.println();
  Serial.println("Starting...");
  
  lcd.begin(16,2);

//before restart:
long tare_offset = LoadCell.getTareOffset(); //save this value to eeprom

//after restart, restore the previous value from eeprom and set:
LoadCell.setTareOffset(tare_offset);

lcd.clear();
  LoadCell.begin();
  float calibrationValue; // calibration value (see example file "Calibration.ino")
  calibrationValue = 456.98; // uncomment this if you want to set the calibration value in the sketch
  //calibrationValue = -24.01; //big scale
  Serial.print("Calibration Value: ");
  Serial.println(calibrationValue);
//#if defined(ESP8266)|| defined(ESP32)
  //EEPROM.begin(512); // uncomment this if you use ESP8266/ESP32 and want to fetch the calibration value from eeprom
//#endif
 //EEPROM.get(calVal_eepromAdress, calibrationValue); // uncomment this if you want to fetch the calibration value from eeprom

  unsigned long stabilizingtime = 2000; // preciscion right after power-up can be improved by adding a few seconds of stabilizing time
  boolean _tare = true; //set this to false if you don't want tare to be performed in the next step
  LoadCell.start(stabilizingtime, _tare);
  LoadCell.setCalFactor(calibrationValue); // set calibration value (float)

    Serial.println("Startup is complete");
    Serial.println(_tare);
 // }
}
 //setting custom values
float fullcan = 353;
//float fullcan = 1360.78;  //empty keg 4525
 float leftincan =0;
 float percentleft=0;
 const int percent =100;

void loop() {
  
  lcd.setCursor(0,0);
  lcd.print("NE IPA      ");
  static boolean newDataReady = 0;     
  lcd.setCursor(0,1);
  lcd.print(percentleft);
  lcd.print("%           ");

  const int serialPrintInterval = 1000; //increase value to slow down serial print activity

  // check for new data/start next conversion:
  if (LoadCell.update()) newDataReady = true;

  // get smoothed value from the dataset:
  if (newDataReady) {
    if (millis() > t + serialPrintInterval) {
      float i = LoadCell.getData();
      Serial.print("Load_cell output val: ");
      Serial.print(i);
      Serial.println("grams");
      newDataReady = 0;
      float currentcan = LoadCell.getData();
      leftincan = ((currentcan/fullcan));
      Serial.print("Full Can: "); 
      Serial.println(fullcan);
      Serial.print("Current Can: ");
      Serial.print(currentcan);
    
     percentleft=(leftincan*100);
      Serial.println(percentleft);
      Serial.print("Percent Remaining: ");
      Serial.println(percentleft);
 
      t = millis();
    }
  }  }

olkal, 2021-02-12:
Thats a useful scale :)
Okay, I think the problem is that you're not actually saving or retrieving anything to/from the EEprom, the library will not do that for you.
The revised code below should give you an idea (I did not test the code, but hopefully it works). Send 't' from the terminal window to save the tare value in EEprom.

```#include <HX711_ADC.h>
#include <EEPROM.h>
#include<LiquidCrystal.h>
LiquidCrystal lcd(12, 11, 5, 4, 3, 2);

//pins:
const int HX711_dout = A1; //mcu > HX711 dout pin
const int HX711_sck = A5; //mcu > HX711 sck pin

//HX711 constructor:
HX711_ADC LoadCell(HX711_dout, HX711_sck);

//const int calVal_eepromAdress = 0;
const int tareOffsetVal_eepromAdress = 4;
unsigned long t = 0;

void setup() {
  Serial.begin(57600);
  Serial.println();
  Serial.println("Starting...");

  lcd.begin(16, 2);

  //restore the previous zero offset value from eeprom and set:
  long tare_offset = 0;
  EEPROM.get(tareOffsetVal_eepromAdress, tare_offset);
  LoadCell.setTareOffset(tare_offset);

  lcd.clear();
  LoadCell.begin();
  float calibrationValue; // calibration value (see example file "Calibration.ino")
  calibrationValue = 456.98; // uncomment this if you want to set the calibration value in the sketch
  //calibrationValue = -24.01; //big scale
  Serial.print("Calibration Value: ");
  Serial.println(calibrationValue);
  //#if defined(ESP8266)|| defined(ESP32)
  //EEPROM.begin(512); // uncomment this if you use ESP8266/ESP32 and want to fetch the calibration value from eeprom
  //#endif
  //EEPROM.get(calVal_eepromAdress, calibrationValue); // uncomment this if you want to fetch the calibration value from eeprom

  unsigned long stabilizingtime = 2000; // preciscion right after power-up can be improved by adding a few seconds of stabilizing time
  boolean _tare = false; //set this to false if you don't want tare to be performed in the next step
  LoadCell.start(stabilizingtime, _tare);
  LoadCell.setCalFactor(calibrationValue); // set calibration value (float)

  Serial.println("Startup is complete");
  Serial.println(_tare);
  // }
}
//setting custom values
float fullcan = 353;
//float fullcan = 1360.78; //empty keg 4525
float leftincan = 0;
float percentleft = 0;
const int percent = 100;

void loop() {

  lcd.setCursor(0, 0);
  lcd.print("NE IPA ");
  static boolean newDataReady = 0;
  lcd.setCursor(0, 1);
  lcd.print(percentleft);
  lcd.print("% ");

  const int serialPrintInterval = 1000; //increase value to slow down serial print activity

  // check for new data/start next conversion:
  if (LoadCell.update()) newDataReady = true;

  // get smoothed value from the dataset:
  if (newDataReady) {
    if (millis() > t + serialPrintInterval) {
      float i = LoadCell.getData();
      Serial.print("Load_cell output val: ");
      Serial.print(i);
      Serial.println("grams");
      newDataReady = 0;
      float currentcan = LoadCell.getData();
      leftincan = ((currentcan / fullcan));
      Serial.print("Full Can: ");
      Serial.println(fullcan);
      Serial.print("Current Can: ");
      Serial.print(currentcan);

      percentleft = (leftincan * 100);
      Serial.println(percentleft);
      Serial.print("Percent Remaining: ");
      Serial.println(percentleft);

      t = millis();
    }
  }

  // receive command from serial terminal
  if (Serial.available() > 0) {
    char inByte = Serial.read();
    if (inByte == 't') refreshTareOffsetValueInEEprom();
  }
}

// zero offset value, calculate and save to EEprom:
void refreshTareOffsetValueInEEprom() {
  long _tare_offset = 0;
  LoadCell.tareNoDelay(); // calculate the new tare / zero offset value
  _tare_offset = LoadCell.getTareOffset(); // get the new tare / zero offset value
  EEPROM.put(tareOffsetVal_eepromAdress, _tare_offset); // save the new tare / zero offset value to EEprom
  LoadCell.setTareOffset(_tare_offset); // set value as library parameter (next restart it will be read from EEprom)
  Serial.print("New tare offset value:");
  Serial.print(_tare_offset);
  Serial.print(", saved to EEprom adr:");
  Serial.println(tareOffsetVal_eepromAdress);
}

Edward-Sams, 2021-02-12:
Thanks olkal!
Tested the code out.  When it starts up, it reads a value of 5353%, after send t, it resets to 0%, which is correct.  However, after the power is unplugged\replugged in, it goes back to the 5353% instead of 0 without a can, or 100 with a can on the scale.

I'm thinking the code is writing to the eeprom, but not retrieving it when booting up.

olkal, 2021-02-12:
I did some testing now, replace `LoadCell.tareNoDelay();` with `LoadCell.tare();` (in line 106 or so).

Edward-Sams, 2021-02-14:
Thank you!  It's working great now.

closed 2021-02-14 by Edward-Sams