olkal/HX711_ADC #64: Drifting Values

kinnif, 2021-02-17:
Hello,

I am using this library for ESP32 and the values drifted to negative 2 (-2.63) over 30 minutes. Is there any possible solutions to fix the drifting issues? Sometimes it will gain additional value but I am wondering if there is a legitimate way to deal with the offset.

Thanks.

olkal, 2021-02-17:
Hi!
Drift or creep is a known problem with load-cells and unfortunately there is no quick fix for it that I know of. 

My understanding is as follow:
- Drift is mostly caused by temperature variations in the load-cell body and in the HX711 chip itself
- Creep is caused by (large) changes in mass on the load-cell and as result the material deflects and continue to deflect a tiny amount further before it more or less stabilizes itself after some time

The following measures could provide some improvement:
- Get a better more "high end" load-cell
- Make some sort of temperature compensation
- I think (not sure) selecting a load cell that is close to the range that you will use it for is better with respect to Drift, as variations in the differencial voltage will have less impact on the output, e.g. don't use a 10kg load-cell if you dont need to measure more than 2kg

You will find more info about this online.

kinnif, 2021-02-18:
Thanks for the response! Was wondering if you have any papers to refer to for both drift and creep?
Also, for 'temperature compensation', do you know of any formulas circulated in the forum?
And looks like I am using a 10kg load cell to detect something light and small which isn't advised in point 3.

Thanks so much for helping :)

olkal, 2021-02-19:
Sorry but I don't have any particular references on this, I don't use load cells for applications where drift/creep over time is of any real consequence. But you are not alone in having this problem, so if you search online you will find lots of information.

amotl, 2021-02-19:
Hi there,

at Hiveeyes, we do have an application where drift, creep and temperature compensation is much of a concern to us because the scale is "always loaded forever" ;]. You might enjoy our discussions at [1,2,3].

While some contents are in German, you will be able to use a convenient translation feature after signing up to the forum.

With kind regards,
Andreas.

[1] https://community.hiveeyes.org/search?q=drift
[2] https://community.hiveeyes.org/search?q=creep
[3] https://community.hiveeyes.org/search?q=compensation

cc @ClemensGruber

kinnif, 2021-02-26:
Many thanks for sharing! Will have a look at it :)

closed 2021-02-26 by kinnif