olkal/HX711_ADC #65: Electrical interference issues when using 4 load cells?

Edward-Sams, 2021-03-10:
Hello World.  I've been working on creating a simple scale using 4 load cells.  I started with using just 1 load cell to make sure my wiring, connections and software were all correct.  Everything is fine.  When I scaled up the project to 4 load cells, I am getting a lot of drift in the value.  I have tried twisting the wires which may have helped a little, but the problem persists.  I have also noticed, just getting my hand near the scale (within 12 inches) the values drift must faster, and a much larger range.   I believe I have some electrical interference.  How do I solve for this?

My code, I am just using the HX711_ADC calibration software.
Power source is a usb connected to my mac (same on I used for the single load cell test).
I have tried soldering my wires directly to the microcontroller and using a dupont connector.

[image]

olkal, 2021-04-18:
Hi, sorry for late response. 
Unfortunately I don't understand what's cousing the problem, it seems to me you have already tried what I would normally have suggested. The loadcell signal is in the millivolt range so quite senitive to electrical noise, but I can't remember that I have seen any difference if I just move my hands up close...

Edward-Sams, 2021-04-19:
Thanks.  I think my issue is more the soldering connection on the hx711 and not the wires.  I'm going to try a different style hx711 this weekend to see if that resolves my issue.  As it works now, it works fine but if i move the setup at all, it will stop working correctly.

olkal, 2021-07-03:
I’m closing this issue because it has been inactive for some time.
Please reopen if you still encounter this issue with the latest version.
Thank you!

closed 2021-07-03 by olkal