olkal/HX711_ADC #66: How to measure average value with HX711_ADC library!

Yatinbaluja, 2021-03-26:
Hi All, 

I am using this library to measure weight with a jar as the values are not constant, it is fluctuating so I want to measure average value of the jar and send it to AWS. Also, I want to measure the average value between 500ms and 1s after that the last value will be send to cloud. I don't care about the samples, how many samples it will take between 500ms and 1s. 

Any help or suggestions would be appreciated!

Thanks,
Yatin

olkal, 2021-04-18:
Hi! 
The library is calculating an average value already. You can increase the number of samples in tha dataset by changing the value of `SAMPLES` in the config.h file. Also be sure to set your HX711 to the 10 SPS setting (ref data sheet).

olkal, 2021-05-31:
I’m closing this issue because it has been inactive for some time.
Please reopen if you still encounter this issue with the latest version.
Thank you!

closed 2021-05-31 by olkal

Yatinbaluja, 2021-06-10:
Hi @olkal I will test it and update you!