olkal/HX711_ADC #67: Remove tare during start up.

KarthikBharadwaj97, 2021-04-09:
Hi, 

Firstly, the library is amazing. Things are really simple and makes much sense to me after using this lib. So I'm stuck with an issue here. I'm using the example code to read the data continuity from hx711. Each time the micro controller is restarted the value is tared and starts from zero and that does not correspond to my input mv(obviousily).
In my application, i have to read and record the data every hour( the weight may increase or decrease). And during the process if the microcontroller is restarted, the value i'm reading starts from zero(which should not happen). I tried to solve this by making the Boolean_tare = false in the example, but the value reading is very high. i have tried figuring out the reason for it but i might be wrong about it. 
       
boolean _tare = false; 
LoadCell.start(stabilizingtime, _tare);

Any help would be greatly appreciated. 
Thanks & Regards
Karthik

jfjoubert, 2021-04-17:
Hi,

I'm also running into this issue. I have a microcontroller with an HX711 loadcell amp which is connected to 4 load cells on an agricultural feeding wagon. It happens quite often that after unloading not all feed is unloaded and the scale is switched off for the night. Then when the scale is switched back on again the existing weight still in the wagon should be displayed, and not the "tared" zero weight. 

I also really like this library, so it would be super nice if the tare at startup could be removed/changed so the HX711 could provide the actual weight which is on the scale at startup. (after calibration obviously).
Thanks and regards,
Jean

olkal, 2021-04-17:
Hi (both of you)!

If you use `LoadCell.start(stabilizingtime, false);`,  the tare function will not run. However, the zero-offset value must be known to get correct output. So, you need to first run the tare function and save the zero-offset value to eeprom. Then restore this value from eeprom after a each reset/restart.
 
I have demonstrated how to do this in example file Persistent_zero_offset.ino. Please try out the example and let me know if it solved the issue or if you still have trouble. (Note that the example will not work as it is on Arduino Due and Zero boards due to lack of eeprom and non-standard eeprom emulation on these boards).

BTW: after a reset/reset the data set will be emty, so you need to let it fill up before you can take a representative reading. The functions `refreshDataSet();` or `getDataSetStatus();` can be used.

jfjoubert, 2021-04-17:
Hi Olav,

Thanks for the info. 
I did take a look at the Persistent_zero_offset sketch, but it didn't work for me. Whenever I do the refreshOffsetValueAndSaveToEEprom function I get a calculated offset value of +- 8557599. When I restart the controller the value read from EEPROM is only 1178. After spending some time I realised that in your refreshOffsetValueAndSaveToEEprom function there is no commit after EEPROM.put if ESP8266 or ESP32 is defined. I added the "if defined" check and the commit and now the correct value is saved to EEPROM for offset and I can restart my controller and the current weight on the scale is still displayed correctly. 
I'm not a C++ expert so I will leave it to you to see if this might have been the issue for me.

My next problem is going to be that whenever a re-calibration is performed, then I will have to recalculate the offset again. It would be nice if I can perform all of this in one go. 
So my question is whether it will be possible to perform a calibration with a known weight and then before rebooting use the newly calculated CalFactor and then also perform the refreshOffset taking into account that the calibration weight is still going to be on the scale? I need to make it as easy as possible for the end-user so a single-step calibration and offset calc would be very nice.

Thanks for spending the time to assist with this.
Regards,
Jean

olkal, 2021-04-18:
Hi, yes I seem to have forgot to include the commit part in the example, I will make an update on that.
Regarding your question: You must first have the correct zero-offset value before you can do a calibration, so I don't understand why you need to recalculate the offset after the calibration? 
Also keep in mind that re-calibration is normally not needed, but the zero-offset will drift with temperature variations and is the value you normally want to re-set.

jfjoubert, 2021-04-18:
Hi, thanks for the info.
I just made the assumption that the zero-offset value will not be calculated correctly without calibrating first. I will try it as suggested and see if I can get it to work in this manner.
Thanks again.
Jean

olkal, 2021-04-18:
Yeah, it's the oher way around: calibration will not be calculated correct unless zero-offset is set first. You're welcome.

olkal, 2021-05-31:
I’m closing this issue because it has been inactive for some time.
Please reopen if you still encounter this issue with the latest version.
Thank you!

closed 2021-05-31 by olkal