olkal/HX711_ADC #68: startMultiple with Samples 64

Comsystems, 2021-05-03:
Hi Olav,
thanks for your great work !
I try to set up a project with multiple load cells and the startMultiple option of your library. Everything works fine when I use standard 16 SAMPLES or no tare option. But when I use 64 samples and tare-option the setup of your example code (Read_2x_load_cell) crashes. Perhaps an memory access issue ?
Can You have a look for this issue ?

Jan

olkal, 2021-05-05:
Hi, and thanks! 
The example file with 2 pcs HX711's connected and 64 SAMPLES should run fine on most microcontrollers, I just tested on an Arduino Uno and it seems to run compile and run without issues. What kind of microcontroller do you have? And how does it crash, will it start at all?

olkal, 2021-07-03:
I’m closing this issue because it has been inactive for some time.
Please reopen if you still encounter this issue with the latest version.
Thank you!

closed 2021-07-03 by olkal