olkal/HX711_ADC #69: no matching function for call to 'HX711_ADC::start(long unsigned int&, boolean&)'

pchatill, 2021-05-04:
Hi.
Thanks for your job!
I want to use the calibration.ino file with a CubeCell HTCC-AB02S board with your library HX711_ADC-1.2.7
But I have the issue on Line 44: LoadCell.start(stabilizingtime, _tare);
"no matching function for call to 'HX711_ADC::start(long unsigned int&, boolean&)'"
Any idea?
Thanks a lot!

olkal, 2021-05-05:
Hi!
Quick answer is that the board is not supported. That said, I'm not sure why it won't compile. 
I tried to install the cubecell package in Arduino board manager to do some testing but I get an error msg: `Error downloading https://resource.heltec.cn/download/package_CubeCell_index.json` ...

olkal, 2021-07-03:
I’m closing this issue because it has been inactive for some time.
Please reopen if you still encounter this issue with the latest version.
Thank you!

closed 2021-07-03 by olkal