olkal/HX711_ADC #70: Sampling rate is higher than specification, check MCU>HX711 wiring and pin designations

Barabba11, 2021-05-05:
Hi! I've just bough this module, I don't know why it measures always zero. I've read quite everywhere, I've tried to invert A+ A-, put a weight before startup, etc. No way to have different result. I can see something if I connect to B+ B- but this is not proportional to weight.
How can I solve my problem? What the author refersr with "wiring and pin"? I've connected in the right way, how can I debug HX711 hardware? Which values of volts shoudl I expect do see on A+ A-? There is any test program that just let me see the 24 bit acquired from the serial protocol?

THis is what I have:
Starting...
Startup is complete
Calibration value: 696.00
HX711 measured conversion time ms: 0.68
HX711 measured sampling rate HZ: 1470.59
HX711 measured settlingtime ms: 12
Note that the settling time may increase significantly if you use delay() in your sketch!
!!Sampling rate is higher than specification, check MCU>HX711 wiring and pin designations
Load_cell output val: 0.00
Load_cell output val: 0.00

Barabba11, 2021-05-05:
CHecking datasheet I see the A+ A- shoud be at the edges of resistor bridge, as well as + and -, on my sensor the story was different, respecting the colours was unbalanced so I need to switch the red with green wire to have a banced bridge.
Doing it produces me some results, but they quite random and not related to the weight itself.
Starting...
Startup is complete
Calibration value: 696.00
HX711 measured conversion time ms: 0.68
HX711 measured sampling rate HZ: 1479.29
HX711 measured settlingtime ms: 12
Note that the settling time may increase significantly if you use delay() in your sketch!
!!Sampling rate is higher than specification, check MCU>HX711 wiring and pin designations
Load_cell output val: 0.00
Load_cell output val: -186.76
Load_cell output val: 10.94
Load_cell output val: 1.38
Load_cell output val: 3.22
Load_cell output val: 0.64
Load_cell output val: -1.20
Load_cell output val: -0.83

How can I change the sampling rate? on my HX711 the pin 15 is grounded, means 10 samples/s, and pin 14 too! means "use on-chip oscillator", Now this internal oscillator is programmable via software?
I couldn't imagine was so hard to have some resilts..
Thanks for help

Barabba11, 2021-05-05:
Using another sketch it works )

Barabba11, 2021-05-05:
I don't know it gaves this error, with other sketch it works good

closed 2021-05-05 by Barabba11

samufnasc, 2025-02-19:
hi, did you solve the problem?

samufnasc, 2025-02-19:
I'm having the same issue

olkal, 2025-02-20:
A sample rate that is far above what is possible for the HX711 is a strong indication that the serial communication between the MCU and the HX711 module is not working. As the message says, check MCU>HX711 wiring and pin designations. 

The person that opened this issue originally seems to have misunderstood and tried to check the HX711>Loadcell wiring and also the HX711 oscillator options, this will not help at all. But he closed the issue before I got a chance to respond.