olkal/HX711_ADC #71: Calibration on Arduino nano: Load_cell output val: nan

Barabba11, 2021-05-05:
Hi again, my phone weight only 50 grams so I need calibration, probably the sketch for calibrate is made for working only under ESP8266, when I decide to write to eeprom I have this:
Load_cell output val: nan (it fills my terminal without stopping)
Can you please help me fix it?

Barabba11, 2021-05-05:
I forgot to change pin 4 and 5 with the default 2 and 3, sorry bu tusually all sketch are based on them. Works fine! Thanks

closed 2021-05-05 by Barabba11

Barabba11, 2021-05-05:
My best congratulation for this library, simply fantastic. Calibration is wonderful, I've tested it with a profesional kitchen scale and the difference was about 1g for heavy and light things

olkal, 2021-05-05:
Many thanks, glad you like it :)

YPayal, 2022-10-05:
I am facing the same issue, I am getting "Load_cell output val: NAN " my connections are perfectly fine. Can someone help me with this.