olkal/HX711_ADC #74: About measurement error by moment

viper0078, 2021-05-31:
Hi there.
I want to make a measuring instrument for 0~500g (to the first decimal place).
When I use one beam type load cell (1kg MAX), I can't measure accurately because the error occurs depending on the position where the weight is applied.
Therefore, I am thinking of using two load cells to avoid the influence of moment.

I have a few questions. How should I calibrate the load cells?
Every single unit?
If so, how should I take into account the error caused by the moment?
Thank you very much.

olkal, 2021-05-31:
Hi!
The load cells I use myself don't seem to change in value if I change the position where I put the load, as the full wheat bridge load cell somewhat compensates for bending moment. But possibly the design of your loadcells are different, or perhaps you apply some twisting moment? 
Sure, you can hook up 2 loadcells to 2 pcs HX711. Calibration will then best be done seperately, and when in use you can just add both values into a total value. 
Another approach is to use 4 pcs 3-wire loadcells hooked up to 1 pcs HX711, this is the usual setup for bathroom scales.

viper0078, 2021-06-19:
Thanks for the comment.

closed 2021-06-19 by viper0078