olkal/HX711_ADC #75: Nonlinear behaviour 4 loadcells ESP 32

Snorreff, 2021-06-05:
Hi,
I have been struggling for a while trying to get a 4 x 50 kg weight, hx711 to run on a Adafruit Huzzah 32. I am using the hx711_adc library and the provided example sketches for calibration and read. Calibration works fine and the system reads 0 kg and the 120 kg calibration weight correct after calibration, however it goes bananas in between, with very high readings (200) with just a few kilos load. I have been testing exactly the same weight and sketches on an Arduino Uno and it works as expected.  This should eliminate quite a lot of potential sources. 
Anyone that have ideas to where I should start looking?
Thanks

olkal, 2021-06-08:
Hi!
Only difference I can think of is that the ESP is a much faster MCU than the Uno, however, the ESP32 is known to work fine provided that the `SCK_DELAY` is enabled in config.h. Could be related to issue [#65](https://github.com/olkal/HX711_ADC/issues/65), but probably not. 
Sorry, but I don't have any usefull ideas...

Snorreff, 2021-06-20:
Hi,
The other difference is 3V versus 5V. Thanks anyhow for your reply!

dsafak, 2021-06-24:
I had this happen to me with 3xhx711s at 144sps on 32u4 with a slightly modified version with this library to allow direct port manipulation.
The issue was speed at which I set clock high and low. Enabling the 1MS delay for ESP in config actually solved the issue for me.

Snorreff, 2021-07-13:
Problem solved. Andreas Spiess is addressing a similar problem in his YouTube video #161
Used

#include "soc/rtc.h"

and 

rtc_clk_cpu_freq_set(RTC_CPU_FREQ_80M);

in the setup to reduce the CPU frequency to 80MHz. 
Finally, two months of trouble shooting paid of 😊
Thanks to both of you!

olkal, 2021-07-14:
Hi, thanks for the info, and good to hear that you solved it!
However I'm puzled that you needed to reduce the core frequency. Did you try to enable `SCK_DELAY`? 
In Andreas Speiss video #161 he is using a completly different HX711 library that use the `shiftIn ` function, hence delay will not work to correct the SCK pulse timing. 
For the HX711_ADC library the SCK pulse timing can be extended by 3 microseconds by enableling `SCK_DELAY` in the config.h file. This has been tested and confirmed to work with the ESP32. See also #35.

Snorreff, 2021-07-17:
Olav,
During my internet search trying to find out how to edit libraries (config.h) I found Andreas’s video. For an amateur facing such an issue, spending hours checking wiring, soldering, building 3 scales using different versions of the Hx711 and three dev. boards I was more than happy finally getting the scale to work. Part of the story includes finding several stupid mistakes, including not allowing the sensors room to bend when testing. 
However, thanks for pushing further. I figured out how to edit the config.h, enable SCK_DELAY and just tested it successfully. Obviously, a better solution  😊

olkal, 2021-07-20:
Excellent! Thanks for confirmation!

closed 2021-07-20 by olkal