olkal/HX711_ADC #77: can it be compatible with hx712 too?

Barabba11, 2021-07-02:
Hi mate, this is not a bug, sorry to create one for this request.
My friend asked me to repair his scale, all is working except processor! It's amazing.. what a chance ))
Since I'm in love with your library I would use this too.. but the board has a hx712, can it be compatible?
Thank you

olkal, 2021-07-02:
Hi!
I'm not sure. The HX712 has a slightly different serial protocol, but using the HX711 library with gain 128 (default setting) it seems like it might work. The MCU will then clock out 25 CLK pulses coresponding to a gain of 128 and rate of 10SPS on the HX712. Give it a try!

Barabba11, 2021-07-05:
Thank you friend, I'll try then as soon I can and report here. 
So if I got it right, I have just to leave the defualt settings and teorically it will work

olkal, 2021-07-06:
Yes, default library settings. The assumption that it might work is just based what I learned from a quick look at the HX712 data sheet and it would be interesting to know the result if you test it, so your feedback is appreciated :)

olkal, 2021-10-03:
I’m closing this issue because it has been inactive for some time.
Please reopen if you still encounter this issue with the latest version.
Thank you!

closed 2021-10-03 by olkal

Barabba11, 2021-10-03:
Dear friend, yes sure. I haven't have time yer to test, when I'll I'll report here ) Thanks