olkal/HX711_ADC #78: HX711 with Blynk

papaduane, 2021-07-20:
When I add
//HX711 constructor (dout pin, sck pin)
HX711_ADC LoadCell_1(HX711_dout_1, HX711_sck_1); //HX711 1
HX711_ADC LoadCell_2(HX711_dout_2, HX711_sck_2); //HX711 2
HX711_ADC LoadCell_3(HX711_dout_3, HX711_sck_1); //HX711 1
HX711_ADC LoadCell_4(HX711_dout_4, HX711_sck_2); //HX711 2

Blynk fails to start correctly

----------------

With out it

start ing v print1
blynk start 2
[19] 
    ___  __          __
   / _ )/ /_ _____  / /__
  / _  / / // / _ \/  '_/
 /____/_/\_, /_//_/_/\_\
        /___/ v1.0.0 on Arduino Nano

[626] Connecting to House
[3813] AT version:1.1.0.0(May 11 2016 18:09:56)
SDK version:1.5.4(baaeaebb)
compile time:May 20 2016 15:08:19
OK
[9097] +CIFSR:STAIP,"192.168.4.116"
+CIFSR:STAMAC,"c4:5b:be:4d:55:c1"
[9105] Connected to WiFi
[19538] Ready (ping: 38ms).
blynk end 2
BOOT_MESSAGE
Blynkstart
blynk end
Blynkstart
blynk end

----------------
with it  in

start ing v print1
blynk start 2
[19] 
    ___  __          __
   / _ )/ /_ _____  / /__
  / _  / / // / _ \/  '_/
 /____/_/\_, /_//_/_/\_\
        /___/ v1.0.0 on Arduino Nano

[626] Connecting to House
[3813] AT version:1.1.0.0(May 11 2016 18:09:56)
SDK version:1.5.4(baaeaebb)
compile time:May 20 2016 15:08:19
OK
[4891] Failed to enable MUX
[14089] +CIFSR:STAIP,"192.168.4.116"
+CIFSR:STAMAC,"c4:5b:be:4d:55:c1"
[14098] Connected to WiFi
[24506] Ready (ping: 25ms).
[50309] Ready (ping: 23ms).
[76317] Ready (ping: 24ms).
[102120] Ready (ping: 26ms).
[127938] Ready (ping: 26ms).
[153801] Ready (ping: 36ms).
[179592] Ready (ping: 25ms).
[205393] Ready (ping: 24ms).

// added

[... 304 lines cut ...]

// Blynk.virtualWrite(V5,a);

terminal.println("iii");

// terminal.println(LoadCell.getData());

terminal.println("jjjj");
 

terminal.println("DOUT2");
Blynk.virtualWrite(V6, "DOUT2");
terminal.println("DOUT3");
Blynk.virtualWrite(V7," DOUT3");

  Serial.println("starting v print c");

delay(1000);

  

  }
*/

Sorry for all the extra lines

olkal, 2021-07-20:
Hi!
First a tip on posting code; if you insert the code using the `<>` button it's easier to read.
Then your issue: 
You can't use pin number 6-11 on the ESP8266, these are already in use by the SPI Flash memory. It looks like you don't have enough available pins for your project with the ESP8266, you might be better off using the ESP32 .

papaduane, 2021-07-20:
Will do.  I am using the ardunio nano with a esp 01 with the “at” code

olkal, 2021-07-20:
Oh, I didn't get that you are running the code on a Nano, sorry. 
The error code that you get seems to be wifi related, but I'm not familar with Blynk or running the ESP8266 via AT code tough. 
However, I can't see any reason why including the HX711 constructors in it self should cause this, it's different  pins and there are no interrupts involved. You have used the same SCK pins (1 & 2) for HX711 no 1, 2, 3 and 4, but again, this will not make any errors on the ESP8266 side.

olkal, 2021-10-03:
I’m closing this issue because it has been inactive for some time.
Please reopen if you still encounter this issue with the latest version.
Thank you!

closed 2021-10-03 by olkal