olkal/HX711_ADC #79: Uploading sketch hangs

dsafak, 2021-07-21:
Hi,
@olkal  if you remember, I had an issue with 144hz hx711s and I ended up using direct port manipulation and a modified version of the library. Device is arduino leonardo, or rather Atmel Atmega 32u4 with caterina bootloader.

I decided to update the library and insert my small modifications back, all worked fine.
BUT,
When I try to upload sketches now, the upload hangs and I either have to go into bootloader manually or need to use ISP to re-program.

I encounter this issue time to time when I am programming.
My experience with the issue and what I think causes it:
 It is usually happens because of either a real fast serial print that bugs out uploading, 
a charachter in serial print that bugs uploading,
not enough memory (dynamic or static, for static when I go over 90%, this bug usually happens) 
or
it happens when I go over 90% flash memory and this bug usually happens.

Right now I have quite good amount of headroom.
Sketch uses 20970 bytes (73%) of program storage space. Maximum is 28672 bytes.
Global variables use 1892 bytes (73%) of dynamic memory, leaving 668 bytes for local variables. Maximum is 2560 bytes.

so this looks to be dynamic memory related to me, but my debugging skills are limited and not sure if I can debug dynamic memory without the atmel tool which I don't have.

Now, about the library and why it happens after I updated to latest library version.
I compared changes between my old version and the latest version and started updating old version line by line and testing uploading each time, twice.
Updating old library HX711_ADC.cpp to new version line by line starting from line 1:
When I add "calFactorRecip = 1/calFactor;" (hx711_ADC.cpp line 226), this bug starts to happen.

I stop at that line so rest of the cpp still stays in the old version without calFactorRecip, so it isn't even utilized yet.

Any idea? Maybe someone else can point a finger at why does it happen.

dsafak, 2021-07-22:
More info:
This is also intermittent.
I managed to get it to work and upload sketched, and get it not to work and hang.
Right now it works, 10 mins ago it didn't, 10 mins earlier it did, so it really looks like a not enough memory issue to me right now.
600bytes free for dynamic allocation yet it doesn't seem enough of an headroom with this new change time to time.

olkal, 2021-07-22:
Hi!
Even if you see different result when you do changes to the code, I don't think the code or the available free memory is the root couse, but rather the USB/COM-port/Serial connection between the MCU and the computer. `Serial.print` functions may have an effect though, you should be able to check this by first uploading a minimal sketch, then your own sketch.

dsafak, 2021-07-22:
I don't have any active printing serial prints.
Uploading works when I upload blibk, then my sketch with new version of library, then it doesn't work even if I upload blink.
I can upload blibk forever otherwise.

Upload works when I use older version of library, I can upload again and again forever, it breaks after 1 flash of new version of the library.

To work around it, I added that useless if statement, it somehow makes new version also work everytime.

so it is weird but it is to do with the library, a bug is being triggered or it is memory related.

dsafak, 2021-07-27:
Well, I have been trying to figure it out and I couldn't.
There is something wrong with this new version.
Unless I start removing some code from my sketch depending on what code, the upload sketch works or doesn't work.
When I stop my loop by an IF and a serialport input to set it true or false, and set my loop to false so stop all execution and then upload, it works.

The only change is the library version, and it awfully looks like a memory problem to me because it starts working when I start removing some code with bigger memory headroom. 26% headroom looks like it is not enough.

Issue is not solved but we can close this issue, I am unable to find why and still trying to work around it, or I'll better go back to old version. If I find why, I'll let you know.

olkal, 2021-07-27:
Hi, sorry to hear that. (and sorry for lack of response, I have been more or less offline for a few days)
I can't find any bugs in the library.
Could be memory headroom as you suggest. That should not be a problem for sketch upload to other typical mcu's maybe it is for the 32u4. I have to say I still hold that 32u4 and it's peculiar bootloader as the main suspect. Is there a reason that you have to use the 32u4? If not I would have tried a different mcu. I appreciate if you keep posting any future updates on this.

closed 2021-07-27 by olkal

dsafak, 2021-07-27:
I am realy deep into the project, I actually made it run on stm32f103 as well but I better stick with 32u4.
I right now figured out something else, it is also related to my execution speed.
The free memory theory, we can ignore it from now on.
I put a 1 seconds delay to my loop, all works, I remove the delay, upload breaks.
I remove some code, it works again, but it is like I can almost remove any function and return 0 and make it work.

so this is a bug with either arduino libraries, compiler or bootloader.
Because when it is broken, Arduino restarts with 1200Baud but bootloader CDC port doesn't expose itself, and I still see normal CDC com port, from there on, ofc AVRDUDE fails and device is in a weird state.

I can go and open serial monitor, send commands, and I see RX Led flash so it receives data, so it is not frozen.
This is most likely a bug buried deep inside somewhere and somehow I am triggering it with last version + my code.

dsafak, 2021-07-30:
It turns out I was right to suspect some funky stuff going on with memory and
the hang was weird, and wasn't a complete hang.
when com port is open at 1200 baud and closed to enter bootloader.
Device was resetting as expected.
but COM port still stayed same,
Normally it should close, create a new virtual one for bootloader.

Apperantly arduino bootloader for 32u4 and kinds, caterina, has a magic boot key, when device resets on 1200 baud open/clsoe, it goes back into SKETCH first, checks a predefined memory location and determines if it should go to bootloader or not.

so a watchdog reset, goes back to sketch, waits till watchdog timeout, but weird enough they do not block skecth execution.
If you remember, when I stopped loop before uploading, it worked, bingo, I stopped execution, so I stopped the weird timing and memory overwrite before watchdog timeout.

With caterina, It was in an unsafe location and sometimes overwritten accidentally., voila, I was right.
So my 32u4 was resetting but getting stuck in a loop and actually not frozen or anything, just stuck in a loop. I know because I could send data from Serial Port and receive nothing, but I could send.

Apperantly this is supposed to be fixed but doesn't look like in the caterine bootloader to date.
There are some extra code in arduino core to apply the fix for LUFA and NiceHood bootloader or any of sort who tries to fix this, or any new version of LUFA or derivatives.
There is basically extra defines which supposed to detect what kind of bootloader you are using and act accordingly.
New LUFA puts the key at RAMEND-1, so it is secure.

Funny thing is, I searched, searched and couldn't find anything, yet yesterday, accidentally bumped into this on github issues.
These are from 2015, 2017 but looks like it's still around and even someone complaining about why it in 2021.

So indeed this is a bug with arduino-core and caterina.
Arduino-core part took care of it for new bootloaders, but still Leonardo uses or micro uses Caterina.

I didn't have this issue even thought I used way more memory, but looks like this is a memory thing so it is a hit and miss, also explains why it works/not works when I do weird stuff with code.