olkal/HX711_ADC #82: How to preserve weight after power restart?

frobert, 2021-10-01:
I put 1kg of weight on the scale, after a power restart the weight resets to 0 and should show 1kg. 
Scale should show the current weight after power reset in my examle 1kg..

Any idea how to solve the issue?

dsafak, 2021-10-02:
When you restart your device, you are using Tare function again.
When you use Tare, the MCU waits fro stabilization time (default 400ms afaik) and then does a reading.
Then that reading is accepted as "0" point, this is what you are experiencing.

I myself never tried without Tare but you can disable tare, enter a known calibration factor and that way you can prevent this.
The possible issue with this approach would be, readings change slightly with environmental changes, so you might lose accuracy.

Try and see if this is okay for you.

olkal, 2021-10-02:
There is an approach to this in the library example file `Persistent_zero_offset.ino`, with functions for saving and restoring the tare/zero offset value using the internal EEPROM. Basically you just send 't' from the Serial terminal to perform tare function and save the offset to EEPROM. The offset value will be restored from EEPROM at next boot.

frobert, 2021-10-03:
Thanks a lot, will try it...

closed 2021-10-03 by frobert