olkal/HX711_ADC #88: Set Gain to 32 bit

gorogdaniel, 2022-01-02:
Hello,

i have only a question becouse i dont find it in the documentation and i like your code very much. 
To my question, How can i set the gain to 32 bit to can read the channel B?

Thanks very much

matt-pinder, 2022-01-03:
I dont believe this works due to src/HX711_ADC.cpp#L306

This doesnt gather the value for (24 + GAIN), it stops at 24 which is only Channel A.

Happy to have this clarified but I don't understand how this will ever work.

olkal, 2022-01-03:
Hi!
You should be able to use channel B by using function `setGain(uint8_t gain)` with argument value 32.

matt-pinder, 2022-01-07:
Hi Olkal, 

I tried this however the value given for channel B is not the same as the Channel A. 

I. E connect a load cell to channel A and put an object on it, note the weight (for example 10KG)

Connect the load cell to channel B, setGain(32) put the same object on it, the weight is not ouput as 10KG, it was 4KG for example. 

This is not suitable for my application unless I am missing something. 

Thanks, 
Matt

olkal, 2022-01-07:
Yes, as you change from 128bit gain to 32bit gain I guess the output value form B will be about 25% of A. I don't think there are many people that are using the B channel, if you want to use two load cells the best solution is to use two HX711's, one for ea load cell.

olkal, 2022-02-09:
Closed due to low activity.

closed 2022-02-09 by olkal