olkal/HX711_ADC #89: setSamplesInUse();

boromyr, 2022-01-04:
Hi, I'm using your library and I noticed that I can't change the number of samples in use while running the code.
In the Setup I entered:
`LoadCell.setSamplesInUse(64);
delay(1000);
Serial.print(LoadCell.getSamplesInUse());`
And I continue to receive 16 as a number of samples, or those compiled.
I'm trying to increase the number of samples during the Tara, then return them to the default value later.

Small doubt: IGN_HIGH_SAMPLE and IGN_LOW_SAMPLE should only be effective with a minimum number of 4 samples, so as to maintain the 2 intermediate values, in the case of 1 and 2 samples, theoretically could eliminate correct values. It is very difficult to reject data for a data set of less than 6, in these cases it would be more correct to use for example the Chauvenet policy.
For a number of samples> 32 instead, eliminate are the 2 extreme values ​​instead it could be completely indifferent. For a set of 128 for example, removing 2 samples will affect very little, perhaps it would be better to increase these values ​​according to SAMPLES.
I have not tested my ideas, I speak at theoretical level, let me know what nor do you think.

olkal, 2022-01-04:
Hi!
The function `setSamplesInUse()` can only be used to reduce number of samples, if the argument is a higher number then the compiled dataset it will have no effect.
I agree that the ignore outlier feature could be improved for larger sample count and it's possible that I will implement something in the future. However, keep in mind that identifying a higher number of outliers from a large data set of e.g 64 or 128 will require the processing of comparing a lot of values.

boromyr, 2022-01-04:
So in theory, if I compile with 64 samples, can I get to 8, and then switch to 16, or can I only decrease compared to the previous value?

olkal, 2022-01-05:
Yes, the upper limit is the compiled array, and you can both increase and decrease within this range. The function itself is located in HX711_ADC.cpp line number 427.

boromyr, 2022-01-14:
I confirm that `setsamplesinuse ()` works as desired now, imposed a higher limit when filling out and then modifies during execution.

I also wanted to ask you if i can reset the array quickly, since `refreshdataset ()` calls the`update ()`, and then loses time to fill the array again. I wish I could empty the array and fill it with NAN (or similar), so as not to affect the average (fill it with "0" would be worse), whenever I get a sudden weight change. My goal is to speed up reading and don't wait for the average to stabilize.

olkal, 2022-01-16:
Functions for manipulating the array as you describe, is not available in the current version of the library. 
I guess I could implemet new functions that allow you to read and/or set the vaue of each member of the array, something like `getDataS(uint8t index)` and `setDataS(uint8t index)`

closed 2022-03-02 by github-actions[bot]