olkal/HX711_ADC #90: Read 4X load cells

Jlaast, 2022-01-07:
Hello Olkal ,
thank you for making this library available as I have found it very useful. 
I am trying to modify the read 2X load cell script , in order to read 4 load cells individually. I added additional pins  and calibration values, in line with was has been already done. Nevertheless, I am unable to read correct values for load cells 3 and 4 , i.e. the displayed values don't change when a load is placed on the cell. I checked my wiring and its seems to be fine. Attached is the code, please let me know if you have any suggestions. Thank you !

#include <HX711_ADC.h>

//pins:
const int HX711_dout_1 = 2; //mcu > HX711 no 1 dout pin
const int HX711_sck_1 = 3; //mcu > HX711 no 1 sck pin
const int HX711_dout_2 = 5; //mcu > HX711 no 2 dout pin
const int HX711_sck_2 = 4; //mcu > HX711 no 2 sck pin
const int HX711_dout_3 = 7; //mcu > HX711 no 3 dout pin
const int HX711_sck_3 = 6; //mcu > HX711 no 3 sck pin
const int HX711_dout_4 = 9; //mcu > HX711 no 4 dout pin
const int HX711_sck_4 = 8; //mcu > HX711 no 4 sck pin

//HX711 constructor (dout pin, sck pin)
HX711_ADC LoadCell_1(HX711_dout_1, HX711_sck_1); //HX711 1
HX711_ADC LoadCell_2(HX711_dout_2, HX711_sck_2); //HX711 2
HX711_ADC LoadCell_3(HX711_dout_3, HX711_sck_3); //HX711 3
HX711_ADC LoadCell_4(HX711_dout_4, HX711_sck_4); //HX711 4

//const int calVal_eepromAdress_1 = 0; // eeprom adress for calibration value load cell 1 (4 bytes)
//const int calVal_eepromAdress_2 = 4; // eeprom adress for calibration value load cell 2 (4 bytes)

unsigned long t = 0;

void setup() {
  Serial.begin(57600); delay(10);
  Serial.println();
  Serial.println("Starting...");

  float calibrationValue_1; // calibration value load cell 1
  float calibrationValue_2; // calibration value load cell 2
  float calibrationValue_3; // calibration value load cell 3
  float calibrationValue_4; // calibration value load cell 4

  calibrationValue_1 = 104.30; // uncomment this if you want to set this value in the sketch
  calibrationValue_2 = 104.90; // uncomment this if you want to set this value in the sketch
  calibrationValue_3 = -91.83; // uncomment this if you want to set this value in the sketch
  calibrationValue_4 = 102.00; // uncomment this if you want to set this value in the sketch

  LoadCell_1.begin();
  LoadCell_2.begin();
  LoadCell_3.begin();
  LoadCell_4.begin();
  unsigned long stabilizingtime = 2000; // tare preciscion can be improved by adding a few seconds of stabilizing time
  boolean _tare = true; //set this to false if you don't want tare to be performed in the next step
  byte loadcell_1_rdy = 0;
  byte loadcell_2_rdy = 0;
  byte loadcell_3_rdy = 0;
  byte loadcell_4_rdy = 0;
  while ((loadcell_1_rdy + loadcell_2_rdy + loadcell_3_rdy + loadcell_4_rdy) < 2) { //run startup, stabilization and tare, both modules simultaneously
    if (!loadcell_1_rdy) loadcell_1_rdy = LoadCell_1.startMultiple(stabilizingtime, _tare);
    if (!loadcell_2_rdy) loadcell_2_rdy = LoadCell_2.startMultiple(stabilizingtime, _tare);
    if (!loadcell_3_rdy) loadcell_3_rdy = LoadCell_3.startMultiple(stabilizingtime, _tare);
    if (!loadcell_4_rdy) loadcell_4_rdy = LoadCell_4.startMultiple(stabilizingtime, _tare);
  }
  if (LoadCell_1.getTareTimeoutFlag()) {
    Serial.println("Timeout, check MCU>HX711 no.1 wiring and pin designations");
  }
  if (LoadCell_2.getTareTimeoutFlag()) {
    Serial.println("Timeout, check MCU>HX711 no.2 wiring and pin designations");
     }
  if (LoadCell_3.getTareTimeoutFlag()) {
        Serial.println("Timeout, check MCU>HX711 no.3 wiring and pin designations");
  }
   if (LoadCell_4.getTareTimeoutFlag()) {
        Serial.println("Timeout, check MCU>HX711 no.4 wiring and pin designations");
  }
  LoadCell_1.setCalFactor(calibrationValue_1); // user set calibration value (float)
  LoadCell_2.setCalFactor(calibrationValue_2); // user set calibration value (float)
  LoadCell_3.setCalFactor(calibrationValue_3); // user set calibration value (float)
  LoadCell_4.setCalFactor(calibrationValue_4); // user set calibration value (float)
  Serial.println("Startup is complete");
}

void loop() {
  static boolean newDataReady = 0;
  const int serialPrintInterval = 0; //increase value to slow down serial print activity

  // check for new data/start next conversion:
  if (LoadCell_1.update()) newDataReady = true;
  LoadCell_2.update();
  LoadCell_3.update();
  LoadCell_4.update();

  //get smoothed value from data set
  if ((newDataReady)) {
    if (millis() > t + serialPrintInterval) {
      float a = LoadCell_1.getData();
      float b = LoadCell_2.getData();
      float c = LoadCell_3.getData();
      float d = LoadCell_4.getData();
      Serial.print("Load_cell 1 output val: ");
      Serial.print(a);
      Serial.print("    Load_cell 2: ");
      Serial.print(b);
      Serial.print("    Load_cell 3: ");
      Serial.print(c);
      Serial.print("    Load_cell 4: ");
      Serial.println(d);
      newDataReady = 0;
      t = millis();
    }
  }

  // receive command from serial terminal, send 't' to initiate tare operation:
  if (Serial.available() > 0) {
    char inByte = Serial.read();
    if (inByte == 't') {
      LoadCell_1.tareNoDelay();
      LoadCell_2.tareNoDelay();
      LoadCell_3.tareNoDelay();
      LoadCell_4.tareNoDelay();
    }
  }

  //check if last tare operation is complete
  if (LoadCell_1.getTareStatus() == true) {
    Serial.println("Tare load cell 1 complete");
  }
  if (LoadCell_2.getTareStatus() == true) {
    Serial.println("Tare load cell 2 complete");
  }
  if (LoadCell_3.getTareStatus() == true) {
    Serial.println("Tare load cell 3 complete");
  }
 if (LoadCell_4.getTareStatus() == true) {
    Serial.println("Tare load cell 4 complete");
  }

}

olkal, 2022-01-09:
Hi!
I don't have a testrig for 4 loadcells, but the code looks okay to me. Only thing I notice is that in line 49 you should change `<2` to `<4` (as you are using 4 loadcells).  But I don't think that should couse any problems.

Have you tried to switch the pins around, e.g. change wiring for LoadCell_1 to pin 8 and 9 and wiring for LoadCell_4 to pin 2 and 3? If so, what happens, is there still no change in output from LoadCell_4?

burgauss, 2022-07-28:
> Hi! I don't have a testrig for 4 loadcells, but the code looks okay to me. Only thing I notice is that in line 49 you should change `<2` to `<4` (as you are using 4 loadcells). But I don't think [...]

Do we have a feedback from the author of this post?